ap--/python-seabreeze #1: Documentation!

ap--, 2015-02-07:
add docstrings.

aztennenbaum, 2017-07-19:
Hi - I am working on a nanosatellite which will be flying with a Ocean Optics USB4000 spectrometer, and will be using python-seabreeze for onboard science data acquisition (python-seabreze is going to space!) 

In support of this mission, I have written documentation for the spectrometer class, based on the documentation here:

https://oceanoptics.com/api/seabreeze/_sea_breeze_a_p_i_8h.html

In addition, our mission requires the use of the pixel binning feature, which is not currently implemented in python-seabreaze, but should be fairly trivial to add. 

I was wondering if you could take a look at the documentation I have written, and see if it all checks out.

## class Spectrometer:

### To be added:

set_pixel_binning_factor(factor)
    Description:
        This function sets the pixel binning factor on the device.
    Parameters:
        factor - The desired pixel binning factor
    Return:
        none
        
get_pixel_binning_factor(factor)
    Description:
        This function gets the pixel binning factor on the device.
    Parameters:
        none
    Return:
        The current pixel binning factor
        
get_max_pixel_binning_factor()
    Description:
        This function gets the max pixel binning factor on the device.
    Parameters:
        boxcar_width - The width of the boxcar smoothing function to be used. Values are typically 1 to 15.
    Return:
        none

Note: 
self._pixels and self._wavelengths will need to be updated when the
pixel binning factor is set. 

Other than that, these functions follow the same form as as the code
for lamp_set_enable. Just copy and paste all the lamp code, and change
the names

### Current functions:

boxcar_width(boxcar_width)
    Description:
        This function sets width of the boxcar filter in the the device's internal memory if that feature is supported.
    Parameters:
        boxcar_width - The width of the boxcar smoothing function to be used. Values are typically 1 to 15.
    Return:
        none

close()
    Description:
        This function closes the spectrometer attached to the system.
    Parameters:
        none
    Return:
        none

continuous_strobe_set_enable(enable)
    Description:
        This function sets the continuous strobe enable state on the device.
    Parameters:
            enable - A boolean used for denoting the desired state (on/off) of the continuous strobe generator. If the value of enable is nonzero, then the continuous strobe will operate. If the value of enable is zero, then the continuous strobe will stop. Note that on some devices the continuous strobe enable is tied to other enables (such as lamp enable or single strobe enable) which may cause side effect
    Return:
[... 169 lines cut ...]
        none
        
tec_set_temperature_C(set_point_C)
    Description:
        This function sets the target (setpoint) TEC temperature.
    Parameters:
        set_point_C - desired temperature, in degrees celsius. 
    Return:
        none
        
trigger_mode(mode)
    Description:
        This function sets the trigger mode for the specified device. Note that requesting an unsupported mode will result in an error.
    Parameters:
        mode - (Input) a trigger mode (0 = normal, 1 = software, 2 = synchronization, 3 = external hardware, etc - check your particular spectrometer's Data Sheet) 
    Return:
        none
        
wavelengths()
    Description:
        This function gets the wavelengths for the spectrometer
    Parameters:
        none
    Return:
        the wavelengths for the spectrometer

ap--, 2017-07-20:
Hi @aztennenbaum,

wow, very very cool! Is it a cubesat? when will it be launched? is there more info on your mission?

For adding the documentation it would be best if you create a pull request. Then it's easiest for me to comment on everything.

I will be very busy the next 3 weeks, but after that, I'll have some time to help :)

-Andreas

zoomx, 2017-07-20:
Very very much appreciated!!!

ap--, 2019-09-11:
sphinx docs are now on https://python-seabreeze.readthedocs.io/en/latest/

closed 2019-09-11 by ap--