ap--/python-seabreeze #6: python module build/install cython error on raspberry pi / raspbian

frumpel, 2015-03-03:
Not sure what the issue here is ... cython seems unhappy

```
root@raspberrypi:~/python-seabreeze-python-seabreeze-v0.3.4# python setup.py install
Compiling ./seabreeze/cseabreeze/wrapper.pyx because it changed.
Cythonizing ./seabreeze/cseabreeze/wrapper.pyx

Error compiling Cython file:
------------------------------------------------------------
...
DEF SBMAXDBUFLEN = 256

class SeaBreezeError(Exception):

    def __init__(self, message=None, error_code=None):
        cdef const char *cmessage
                       ^
------------------------------------------------------------

seabreeze/cseabreeze/wrapper.pyx:23:24: Syntax error in C variable declaration
Traceback (most recent call last):
  File "setup.py", line 35, in <module>
    ext_modules=cythonize(extensions),
  File "/usr/lib/pymodules/python2.7/Cython/Build/Dependencies.py", line 517, in cythonize
    cythonize_one(pyx_file, c_file, quiet, options)
  File "/usr/lib/pymodules/python2.7/Cython/Build/Dependencies.py", line 540, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: ./seabreeze/cseabreeze/wrapper.pyx
```

frumpel, 2015-03-03:
Additional info (after reading this: https://github.com/cython/cython/wiki/FAQ#how-do-i-use-const): because raspbian is a form of debian and as such barely out of the last century I am tied to cython 0.15

ap--, 2015-03-03:
I'll add the cython-generated C file to the repository, to get rid of the cython dependency.

In the meantime install pip: [what-is-the-official-preferred-way-to-install-pip](http://stackoverflow.com/questions/5585875/what-is-the-official-preferred-way-to-install-pip-and-virtualenv-systemwide)
And pip install cython==0.20.2.

commit c6a0fb06 (2015-03-11): Make cython dependency optional

Closes #6
Provides the cython generated C code for the case where cython is not installed.
closed 2015-03-11 by ap--