ap--/python-seabreeze #7: Error listing the devices

dalonsoa, 2015-03-11:
Hi,

I installed the precompiled seabreeze Windows 7 library from here:

https://github.com/ap--/libseabreeze/releases

and the precompileled windows wheel from here:

https://github.com/ap--/python-seabreeze/releases

Everything installs fine and the hardware seems to be configured correctly by Windows. However, when I try to list the devices connected this is what I get:

    import seabreeze.spectrometers as sb
    devices = sb.list_devices()
    ---------------------------------------------------------------------------
    TypeError                                 Traceback (most recent call last)
    <ipython-input-2-722e25f61c75> in <module>()
          1 import seabreeze.spectrometers as sb
    ----> 2 devices = sb.list_devices()

    seabreeze\cseabreeze\wrapper.pyx in seabreeze.cseabreeze.wrapper.device_list_devices     (./seabreeze/cseabreeze/wrapper.c:3035)()

    seabreeze\cseabreeze\wrapper.pyx in seabreeze.cseabreeze.wrapper.device_get_serial_number   (./seabreeze/cseabreeze/wrapper.c:4002)()

    TypeError: Type str doesn't support the buffer API

I have no idea what this means. I am using Windows 7 (32 bits) with Python 3.4.

Best wishes, 
Diego

ap--, 2015-03-11:
Hmmm, I can't reproduce this with Python3.4 under Linux.
I'll try to test it with windows and report back.

closed 2015-03-11 by ap--

ap--, 2015-03-11:
It'll take a while until the new wheels are available because some appveyor updates broke things ... Just check back every now and then.

dalonsoa, 2015-03-11:
Perfect! It seems that with the newer version of the wheel it works! Thanks!

commit e19bebd7 (2015-03-11): Fix Python3 compatibility in cseabreeze

Fixes #7