ap--/python-seabreeze #10: Windows XP (32 bits): cseabreeze does not load and pyseabreeze does not find devices

dalonsoa, 2015-05-06:
Hi, 

I have followed the instructions of the main page about installing python-seabreeze on Windows ( I am trying to do so in Windows XP (32 bits) ). However, when I run the simple example programs, it systematically says that cseabreeze can not be loaded and that pyseabreeze will be used instead. However, this does not work because the spectrometer is configured using WinUSB and not libusb (if configured with libusb, the proprietary OceanView does not find the spectrometer, which is not convenient; additionally, libusb only gets incomplete information from the spectrometer and python-seabreeze does not work, either!).

This configuration with WinUSB and using cseabreeze works perfect in another PC with Windows 7 (32 bits), so I think there must be an issue with the versions of the libraries or the way seabreeze has been compiled with cython. 

I am using: 
- Python 3.4
- SeaBreezeSetup-x86bit-v3.0.6.msi
- seabreeze-0.4.3-cp34-none-win32.whl

Any suggestion on where the problem might be will be very welcome.

Best wishes

ap--, 2015-05-06:
Can you run:
```
import seabreeze.cseabreeze
```

**if it looks somewhat like this:**
```
import seabreeze.cseabreeze
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-e31549f0abc5> in <module>()
----> 1 import seabreeze.cseabreeze

/opt/localpythonlib/lib/python2.7/site-packages/seabreeze/cseabreeze/__init__.py in <module>()
      3 """
      4 
----> 5 from .wrapper import (SeaBreezeError,
      6                       SeaBreezeDevice,
      7                       initialize,

ImportError: libseabreeze.so: cannot open shared object file: No such file or directory
```

It can't find the shared library. So then you have to add the folder that contains the dll to your path.

**else:**
the error you'll see probably means you don't have the right runtime libraries. You should be able to figure that out by opening seabreeze\cseabreeze\wrapper.dll with this [tool](http://dependencywalker.com/) and it might tell you that there's a msvcpXXX.dll (or something like that) missing. You then have to install the right runtime library (google for _missing msvcpXXX.dll_)

Let me know how it goes.

ap--, 2015-10-20:
closed for now.

closed 2015-10-20 by ap--