ap--/python-seabreeze #11: intensities() returns all pixels for QE65000, not only valid ones

sdickreuter, 2015-05-06:
When checking the length of the returned wavelengths I get 1024 and for the returned intensities I get 1044.
So I guess that seabreeze returns all pixels. This is a problem when one wants to relate the wavelengths with the intensities.
This might be fixed by using the same strategy as in python-oceanoptics, by using a dictionary of the models and a corresponding slice which only returns valid pixels, I  haven't found a way to do this within the seabreeze library.
Does anyone else experience this problem ?

ap--, 2015-05-06:
Which backend?
Can you run the script from the scripts folder for both backends?
```
seabreeze-compare cseabreeze
```
and
```
seabreeze-compare pyseabreeze
```

sdickreuter, 2015-05-07:
Problem solved, my own code was wrong!
I thought that wavelengths() would return 1024 values as it did with python-oceanoptics, but checking with seabreeze-compare it became clear to me that it now returns 1044 values for the wavelength as well as for the intensities.
Thanks!

closed 2015-05-07 by sdickreuter