ap--/python-seabreeze #12: Strange Effect when using python-seabreeze in combination with gtk

sdickreuter, 2015-05-07:
I'm not sure if this would better fit to the bugtracker of gtk, but maybe someone else using python-seabreeze can check if the problem exist on other computers too, and it's not a problem of my configuration.
When calling the following code with python 3.4, wrong values for the wavelength are returned:
```python
from gi.repository import Gtk
import seabreeze.spectrometers as sb

devices = sb.list_devices()
s = sb.Spectrometer(devices[0])
print(s.intensities())
print(s.wavelengths())

```
When commenting out **from gi.repository import Gtk**, the right values are reported.
In both cases the intensities are correct.
I can't get my head around why gtk might interfere with seabreeze ...

ap--, 2015-05-07:
This is very weird...
Does this happen for both backends?
Please provide the printed output.

ap--, 2015-05-07:
Please do:
```
import seabreeze.spectrometers as sb
spec = sb.Spectrometer.from_serial()
for i in range(1, 5):
    print(spec.read_eeprom_slot(i))
```
with and without the Gtk import.

sdickreuter, 2015-05-07:
I also testet the pyseabreeze backend and the problem happens only with the cseabreeze backend.
The scrambled wavelength vector looks like this:
```
[ -6.10800000e+03  -4.42500000e+03  -3.07200000e+03 ...,   6.57228806e+09
   6.59143373e+09   6.61061655e+09]
```
Running your code gives to following result:
```
AttributeError: 'Spectrometer' object has no attribute 'read_eeprom_slot'
```

ap--, 2015-05-08:
Can you test if the newest commit on master fixes this problem?

sdickreuter, 2015-05-08:
Unfortunately the result is the same:
```
[ -6.10800000e+03  -4.42500000e+03  -3.07200000e+03 ...,   6.57228806e+09
   6.59143373e+09   6.61061655e+09]
```
A workaround would be to import gtk after ```s = sb.Spectrometer(devices[0])```, then everything works fine

ap--, 2016-02-13:
Can you still reproduce this issue?
I tried, and I can't anymore.
I tried going back to old libseabreeze versions, python-seabreeze versions and python interpreters.

So my guess is that it might have been a gtk3 bug, that is fixed now? Since this is the only thing I didn't test.

sdickreuter, 2016-02-18:
At the moment I use seabreeze under windows with qt instead of gtk which works fine.
I just tried to get the latest version running under linux to test with gtk, but unfortunately I get the following error:

`>>> import seabreeze.spectrometers as sb

Traceback (most recent call last):
  File "/home/dunkelfeld/PyCharmProjects/python-seabreeze/seabreeze/backends.py", line 30, in _use_cseabreeze
import seabreeze.cseabreeze as sbb
  File "/home/dunkelfeld/PyCharmProjects/python-seabreeze/seabreeze/cseabreeze/__init__.py", line 5, in <module> from .wrapper import (SeaBreezeError, 
ImportError: No module named 'seabreeze.cseabreeze.wrapper'
`

Now I'm a bit confused, do you know why this error occurs ?

ap--, 2016-02-21:
Did you install the module properly?

This happens, if you install the module via ```python setup.py install``` and try to run it from the same path. In your case: ```/home/dunkelfeld/PyCharmProjects/python-seabreeze```

Also you closed the issue. Does this mean you also can't reproduce the bug anymore?

sdickreuter, 2016-02-23:
Thanks for the tip, that was actually the Problem, the new version works perfectly in a different directory.
Unfortunately the bug is still present and the wavelength vector still has the wrong values when importing gtk3 (version 3.18.7-1). Oddly enough, when importing gtk2 from gi.repository everything is fine.
Also sorry for closing the issue, I just clicked the wrong button.

ap--, 2016-02-23:
Can you run the following code and post the output:

```{python}
from gi.repository import Gtk, GObject

import seabreeze.cseabreeze as cseabreeze
import numpy
import sys

print ( sys.version )
print ( cseabreeze.__file__ )
print ( Gtk._version )
print ( GObject.pygobject_version )

device, = cseabreeze.wrapper.device_list_devices()
cseabreeze.wrapper.device_open(device)
specID, = cseabreeze.wrapper.device_get_spectrometer_feature_id(device)
speclen = cseabreeze.wrapper.spectrometer_get_formatted_spectrum_length(device, specID)

out = numpy.empty((speclen,), numpy.double)
cseabreeze.wrapper.spectrometer_get_wavelengths(device, specID, out)

print ( "First ten wavelengths:" )
print ( out[:10] )
```

Thanks

ap--, 2016-02-23:
And also with the Gtk, GObject import after the sys import. :) Thanks.

sdickreuter, 2016-02-24:
No problem,
Gtk first: 
```python
3.5.1 (default, Dec  7 2015, 12:58:09) 
[GCC 5.2.0]
/usr/lib/python3.5/site-packages/seabreeze-0.5.3-py3.5-linux-x86_64.egg/seabreeze/cseabreeze/__init__.py
3.0
(3, 18, 2)
1044
First ten wavelengths:
[ -6.10800000e+03  -4.42500000e+03  -3.07200000e+03  -2.01300000e+03
  -1.21200000e+03  -6.33000000e+02  -2.40000000e+02   3.00000000e+00
   1.32000000e+02   1.83000000e+02]
```

Gtk after sys:
```python
3.5.1 (default, Dec  7 2015, 12:58:09) 
[GCC 5.2.0]
/usr/lib/python3.5/site-packages/seabreeze-0.5.3-py3.5-linux-x86_64.egg/seabreeze/cseabreeze/__init__.py
3.0
(3, 18, 2)
1044
First ten wavelengths:
[ -6.10800000e+03  -4.42500000e+03  -3.07200000e+03  -2.01300000e+03
  -1.21200000e+03  -6.33000000e+02  -2.40000000e+02   3.00000000e+00
   1.32000000e+02   1.83000000e+02]
```
So now the wrong wavelengths are returned regardless of the position of the gtk import.
That new behavior might be related to a package upgrade of gtk3 I recently installed.

Here is the output without importing gtk and gobject:
```python
3.5.1 (default, Dec  7 2015, 12:58:09) 
[GCC 5.2.0]
/usr/lib/python3.5/site-packages/seabreeze-0.5.3-py3.5-linux-x86_64.egg/seabreeze/cseabreeze/__init__.py
1044
First ten wavelengths:
[ 184.34537107  185.15048402  185.95551917  186.76047652  187.56535608
  188.37015785  189.17488183  189.97952804  190.78409646  191.58858712]```

Maybe it is something distribution specific, which distribution did you use for testing ?

ap--, 2021-08-16:
I'm closing this one, because it never got reported again... 🤷

closed 2021-08-16 by ap--