ap--/python-seabreeze #16: Discrepancy between python-seabreeze and python-oceanoptics

yakir12, 2015-10-01:
I'm sure I'm doing something wrong but `intensities` does not result in the same spectrum when I use `python-seabreeze` and `python-oceanoptics`. While the spectrum from `python-oceanoptics` looks like what I expect, the spectrum from `python-seabreeze` looks like weird saturated noise (tons of high values when the spectrum is in complete darkness). 

I'm sure it has to do with differences in the default settings between the two forks: `raw=False, only_valid_pixels=True, correct_nonlinearity=True, correct_darkcounts=True, correct_saturation=True, etc`.

How can I get  `python-seabreeze`'s `intensities` to output something similar to what it does in `python-oceanoptics`? And maybe more importantly, why...?

ap--, 2015-10-01:
Need more info:
* which spectrometer model?
* which backend are you using?

yakir12, 2015-10-01:
Sorry, of course:
* USB2000
* Since I got some errors when trying to install the cseabreeze backend, I think I'm using the pyseabreeze...

When running it (say from Python) I use:
```python
>>> import seabreeze
>>> seabreeze.use('pyseabreeze')
>>> import seabreeze.spectrometers as sb
```

ap--, 2015-10-01:
Okay.
In python-oceanoptics, the only keyword parameter that does anything is: _only_valid_pixels_

So if you want to compare the two, please do (I assume, that **spec** is an instance of the class for your spectrometer):

* **python-oceanoptics** <br/>
  ```spec.intensities(only_valid_pixels=False) / 4095.```
* **python-seabreeze** <br/>
  ```spec.intensities(correct_dark_counts=False, correct_nonlinearity=False)```

Let me know how it goes.

yakir12, 2015-10-01:
I'll report back (Australia) tomorrow, the spec is at work :|

yakir12, 2015-10-01:
It seems like I'm getting values larger than one with `python-seabreeze`.  
* python-oceanoptics:
```python
>>> import oceanoptics
>>> spec = oceanoptics.get_a_random_spectrometer()
>>> y = spec.intensities(only_valid_pixels=False) / 4095
>>> y
array([ 0.        ,  0.04151404,  0.04126984, ...,  0.04249084,
        0.04249084,  0.04224664])
>>> sum(y > 1)
0
```
* pyseabreeze backend in python-seabreeze:
```python
>>> import seabreeze
>>> seabreeze.use('pyseabreeze')
>>> import seabreeze.spectrometers as sb
>>> devices = sb.list_devices()
>>> spec = sb.Spectrometer(devices[0])
>>> y = spec.intensities(correct_dark_counts=False, correct_nonlinearity=False)
>>> y
array([ 3072.,  3240.,  3493., ...,     0.,     0.,     0.])
>>> sum(y > 1)
1024
```

ap--, 2015-10-02:
Hey, it seems i am wrong about pyseabreeze returning normalized spectra. (It's been a while since I worked on this and I have to double check why i made that decision).

And these two lines are wrong: https://github.com/ap--/python-seabreeze/blob/master/seabreeze/pyseabreeze/interfaces/spectrometer.py#L133L134 which is why your spectra don't make sense.

I'll push a patch soon.

yakir12, 2015-10-05:
Hmm... I'm getting this error now:
```python
>>> import seabreeze
>>> seabreeze.use('pyseabreeze')
>>> import seabreeze.spectrometers as sb
>>> devices = sb.list_devices()
>>> spec = sb.Spectrometer(devices[0])
>>> y = spec.intensities(correct_dark_counts=False, correct_nonlinearity=False)
Traceback (most recent call last):
  File "seabreeze/pyseabreeze/interfaces/common.py", line 14, in decorated_func
    return func(*args, **kwargs)
  File "seabreeze/pyseabreeze/interfaces/spectrometer.py", line 134, in get_formatted_spectrum
    tsorted = tmp[idx] & np.array((0xFF, 0x0F)*(N/2))  # high nibble not guaranteed to be pulled low
IndexError: index 4097 is out of bounds for size 4097
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "seabreeze/spectrometers.py", line 148, in intensities
    out[transfered_N:])
  File "seabreeze/pyseabreeze/wrapper.py", line 123, in spectrometer_get_formatted_spectrum
    return device.interface.get_formatted_spectrum(out)
  File "seabreeze/pyseabreeze/interfaces/common.py", line 23, in decorated_func
    raise SeaBreezeError(msg)
seabreeze.pyseabreeze.interfaces.common.SeaBreezeError: Error while reading raw spectrum
```
or this:
```python
>>> y = spec.intensities()
Traceback (most recent call last):
  File "seabreeze/pyseabreeze/interfaces/common.py", line 14, in decorated_func
    return func(*args, **kwargs)
  File "seabreeze/pyseabreeze/interfaces/spectrometer.py", line 134, in get_formatted_spectrum
    tsorted = tmp[idx] & np.array((0xFF, 0x0F)*(N/2))  # high nibble not guaranteed to be pulled low
IndexError: index 4097 is out of bounds for size 4097
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "seabreeze/spectrometers.py", line 148, in intensities
    out[transfered_N:])
  File "seabreeze/pyseabreeze/wrapper.py", line 123, in spectrometer_get_formatted_spectrum
    return device.interface.get_formatted_spectrum(out)
  File "seabreeze/pyseabreeze/interfaces/common.py", line 23, in decorated_func
    raise SeaBreezeError(msg)
seabreeze.pyseabreeze.interfaces.common.SeaBreezeError: Error while reading raw spectrum
```

ap--, 2015-10-06:
Let me know if it's fixed now.

yakir12, 2015-10-07:
I pulled, and went through the [installation](https://github.com/ap--/python-seabreeze#ubuntu-1204-and-probably-all-other-debian-based-distros) for the `pyseabreeze backend` and got this error:
```bash
sudo ./misc/install_udev_rules.sh
# SEABREEZE: copying 10-oceanoptics.rules to /etc/udev/rules.d/
cp: cannot stat ‘10-oceanoptics.rules’: No such file or directory
# SEABREEZE: reloading udev rules
```
went on with the rest of it, and got similar errors when trying to fetch the intensities:
```python
Traceback (most recent call last):
  File "seabreeze/pyseabreeze/interfaces/common.py", line 14, in decorated_func
    return func(*args, **kwargs)
  File "seabreeze/pyseabreeze/interfaces/spectrometer.py", line 135, in get_formatted_spectrum
    ret = numpy.array(struct.unpack("<" + "H"*self._PIXELS, tsorted), dtype=numpy.double)
error: unpack requires a string argument of length 4096
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "seabreeze/spectrometers.py", line 148, in intensities
    out[transfered_N:])
  File "seabreeze/pyseabreeze/wrapper.py", line 123, in spectrometer_get_formatted_spectrum
    return device.interface.get_formatted_spectrum(out)
  File "seabreeze/pyseabreeze/interfaces/common.py", line 23, in decorated_func
    raise SeaBreezeError(msg)
seabreeze.pyseabreeze.interfaces.common.SeaBreezeError: Error while reading raw spectrum
```
Maybe I'm doing something wrong somewhere else? Want me to run some other tests to make sure the problem isn't somewhere else? Let me know.

ap--, 2015-10-08:
I missed a cast to int64. fixed now.

> Maybe I'm doing something wrong somewhere else? Want me to run some other tests to make sure the problem isn't somewhere else? Let me know.

I hope it works now.

yakir12, 2015-10-08:
Yap, that looks good. I think you can close this issue (I'll continue my rant at #17). Thanks!

commit ed18a6e5 (2015-10-05): [pyseabreeze] fix byte sorting for USB2000

* fixes #16
closed 2015-10-09 by ap--