ap--/python-seabreeze #17: How to extract the saturation value?

ap--, 2015-10-01:
Moved from ap--/python-oceanoptics#23 asked by @yakir12:

> Fantastic. I'm assuming the variable I want is _MAX_PIXEL_VALUE.
> I'm a total noob with Python, and I'm doing all of this from Julia with PyCall. Everything works great, but > how do I actually access _MAX_PIXEL_VALUE?
> 
> ```python
> >>> spec._MAX_PIXEL_VALUE()
>   Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   AttributeError: 'Spectrometer' object has no attribute '_MAX_PIXEL_VALUE'
> ```
> Excuse the noobness...

ap--, 2015-10-01:
### short answer:
 You shouldn't, but when using the **pyseabreeze** backend (it's not possible using the **cseabreeze** backend):
```python
spec._dev.interface._MAX_PIXEL_VALUE
```

### long answer:
These values are hardcoded in the seabreeze library (and therefore the cseabreeze backend) and model specific. pyseabreeze uses the same model specific values. When you know your USB product_Id, you know the max_pixel_value. Because of this you do not need to query this during runtime.
python-seabreeze returns spectra normalized to 1. (If it doesn't, consider it a bug.)

yakir12, 2015-10-01:
"python-seabreeze returns spectra normalized to 1" => fantastic!
Thanks a lot!

yakir12, 2015-10-08:
So the whole reasoning behind this issue is the need to know if something got saturated or not. I'm plotting the spectra and want to put a limit to the y-axis. This way a user will know if the integration time is too large. We have a couple of different specs in our lab (and might get others in the future), and I'd like it to work dynamically with "any" spectrometer. 
`spec._dev.interface._MAX_PIXEL_VALUE` works great (for any `Julia` users, the equivalent command is `spec[:_dev][:interface][:_MAX_PIXEL_VALUE]`), but yea, if you meant for the spectra to come out scaled to one, they still don't.

ap--, 2015-11-15:
Small update:

As soon as [this](https://sourceforge.net/p/seabreeze/tickets/5/) gets fully merged, I'll add support for normalizing all spectra to the maximum intensity value.

ap--, 2019-09-11:
possible for all spectrometers with version 1.0.0 :tada: 
```python
In [1]: from seabreeze.spectrometers import Spectrometer                                                 
In [2]: spec = Spectrometer.from_first_available()                                                       
In [3]: spec.max_intensity                                                                               
Out[3]: 65535.0
```

closed 2019-09-11 by ap--