ap--/python-seabreeze #18: isready()

yakir12, 2015-10-12:
It would be an awesome feature to have an `isready` functionality for testing if the spectrometer is ready for input (e.g. via `spec.integration_time_micros(12000)`) and/or output (e.g. `spec.intensities()`).

ap--, 2015-10-12:
All spectrometer methods should be blocking.

What's your specific use case for an ```isready``` method?

yakir12, 2015-10-12:
I want to fetch intensities when the spec is ready and not before by using
asynchronous coroutines. All of this is done in Julia (not Python) with
channels, tasks, and @async.
What do you mean with "All spectrometer methods should be blocking"? Maybe
I can work around it?

ap--, 2015-10-12:
If you call spec.intensities() pyseabreeze sends a request command to the spectrometer and starts reading the spectrum information from the spectrometer.

By default most oceanoptics spectrometers are in a free running trigger mode. This means, that once you set the integration time, the spectrometer is constantly acquiring spectra. When a spectrum is requested via usb the spectrometer is sending the next valid acquired spectrum. This means in free running trigger mode, dependent on when the spectrum is requested, it can take from _\[insert really short time\]_ up to _[insert the integration time you set\]_ until the spec.intensities() call returns.

if you call spec.intensities() in a loop, the loop delay will be the integration time you set (plus some overhead).

yakir12, 2015-10-12:
I'm trying to avoid changing the integration time and fetching the
intensities at the same time. I think it should be possible to do that in
my specific setup but thought it would be great if there was a isready()
function that could tell me when the spec is ready for any kind of
communication.

ap--, 2015-10-12:
I should check if pyseabreeze is threadsafe. In the meantime, if you want to change the integration time and read spectra from two different threads, you may need to implement the thread locking yourself.

I would recommend to use a single thread to communicate with the spectrometer and implement all asynchronous calls on this thread. In the main loop of this thread you can check if other threads want to change the integration time and do the change on demand.

yakir12, 2015-10-12:
Awesome advice. I'll keep tinkering. Thanks for all the work and sorry for
opening all these "issues".