ap--/python-seabreeze #19: python setup.py install not working on linux or os.x

100ideas, 2015-11-11:
Followed instructions in readme to install on raspberrian (armv7l). Everything seemed to compile, but coudln't successefully import the seabreeze package in python.

I was unable to install any of the precompiled wheel packages on os.x, ```pip install seabreeze-0.5.1-cp27-none-macosx_10_5_x86_64.whl``` would report an error like ```seabreeze-0.5.1-cp27-none-macosx_10_5_x86_64.whl is not a supported wheel on this platform,``` but I was able to compile from source. Had the same problem though as on rpi.

```bash
pi@opentrons1 ~/spec $ git clone https://github.com/ap--/python-seabreeze.git
Cloning into 'python-seabreeze'...
remote: Counting objects: 374, done.
remote: Total 374 (delta 0), reused 0 (delta 0), pack-reused 374
Receiving objects: 100% (374/374), 168.01 KiB | 0 bytes/s, done.
Resolving deltas: 100% (216/216), done.
Checking connectivity... done.
pi@opentrons1 ~/spec $ cd python-seabreeze
pi@opentrons1 ~/spec/python-seabreeze $ ls
appveyor  appveyor.yml  LICENSE.md  misc  README.md  scripts  seabreeze  setup.py  travis
pi@opentrons1 ~/spec/python-seabreeze $ sudo python setup.py install
running install
running bdist_egg
running egg_info
creating seabreeze.egg-info
writing pbr to seabreeze.egg-info/pbr.json
writing seabreeze.egg-info/PKG-INFO
writing top-level names to seabreeze.egg-info/top_level.txt
writing dependency_links to seabreeze.egg-info/dependency_links.txt
writing manifest file 'seabreeze.egg-info/SOURCES.txt'
reading manifest file 'seabreeze.egg-info/SOURCES.txt'
writing manifest file 'seabreeze.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
running build_ext
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/seabreeze
creating build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/__init__.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/wrapper.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze
creating build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/interfaces/nonlinearity.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/interfaces/common.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/interfaces/__init__.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/interfaces/communication.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/interfaces/defines.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/interfaces/thermoelectric.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/interfaces/wavelength.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/interfaces/spectrometer.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/pyseabreeze/interfaces/eeprom.py -> build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces
copying build/lib.linux-armv7l-2.7/seabreeze/spectrometers.pyc -> build/bdist.linux-armv7l/egg/seabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/__init__.py -> build/bdist.linux-armv7l/egg/seabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/backends.py -> build/bdist.linux-armv7l/egg/seabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/__init__.pyc -> build/bdist.linux-armv7l/egg/seabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/backends.pyc -> build/bdist.linux-armv7l/egg/seabreeze
creating build/bdist.linux-armv7l/egg/seabreeze/cseabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/cseabreeze/__init__.py -> build/bdist.linux-armv7l/egg/seabreeze/cseabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/cseabreeze/__init__.pyc -> build/bdist.linux-armv7l/egg/seabreeze/cseabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/cseabreeze/wrapper.so -> build/bdist.linux-armv7l/egg/seabreeze/cseabreeze
copying build/lib.linux-armv7l-2.7/seabreeze/spectrometers.py -> build/bdist.linux-armv7l/egg/seabreeze
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/__init__.py to __init__.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/wrapper.py to wrapper.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces/nonlinearity.py to nonlinearity.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces/common.py to common.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces/__init__.py to __init__.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces/communication.py to communication.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces/defines.py to defines.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces/thermoelectric.py to thermoelectric.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces/wavelength.py to wavelength.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces/spectrometer.py to spectrometer.pyc
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/pyseabreeze/interfaces/eeprom.py to eeprom.pyc
creating stub loader for seabreeze/cseabreeze/wrapper.so
byte-compiling build/bdist.linux-armv7l/egg/seabreeze/cseabreeze/wrapper.py to wrapper.pyc
[... 101 lines cut ...]
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv7l-2.7/seabreeze/cseabreeze/wrapper.o -lseabreeze -lusb -o build/lib.linux-armv7l-2.7/seabreeze/cseabreeze/wrapper.so
running build_scripts
creating build/scripts-2.7
copying and adjusting scripts/seabreeze-compare -> build/scripts-2.7
changing mode of build/scripts-2.7/seabreeze-compare from 644 to 755
pi@opentrons1 ~/spec/python-seabreeze $ ls
appveyor  appveyor.yml  build  LICENSE.md  misc  README.md  scripts  seabreeze  setup.py  travis
pi@opentrons1 ~/spec/python-seabreeze $ ls build/
lib.linux-armv7l-2.7  scripts-2.7  temp.linux-armv7l-2.7
pi@opentrons1 ~/spec/python-seabreeze $ python
Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.path.insert(0,'/home/pi/spec/python-seabreeze/build/lib.linux-armv7l-2.7')
>>> import seabreeze.spectrometers as sb
>>> sb
<module 'seabreeze.spectrometers' from '/home/pi/spec/python-seabreeze/build/lib.linux-armv7l-2.7/seabreeze/spectrometers.py'>
>>> sb.list_devices()[0]
<SeaBreezeDevice USB2000PLUS:USB2K10109>
>>> spec = sb.Spectrometer(sb.list_devices()[0])
>>> spec
<Spectrometer USB2000PLUS:USB2K10109>
>>> 
```

100ideas, 2015-11-11:
Also, ```sudo python setup.py develop``` **did** successfully add the package to my system's python modules.

```bash
pi@opentrons1 ~/spec/python-seabreeze $ sudo python setup.py develop
running develop
running egg_info
writing pbr to seabreeze.egg-info/pbr.json
writing seabreeze.egg-info/PKG-INFO
writing top-level names to seabreeze.egg-info/top_level.txt
writing dependency_links to seabreeze.egg-info/dependency_links.txt
reading manifest file 'seabreeze.egg-info/SOURCES.txt'
writing manifest file 'seabreeze.egg-info/SOURCES.txt'
running build_ext
copying build/lib.linux-armv7l-2.7/seabreeze/cseabreeze/wrapper.so -> seabreeze/cseabreeze
Creating /usr/local/lib/python2.7/dist-packages/seabreeze.egg-link (link to .)
Removing seabreeze 0.5.1 from easy-install.pth file
Adding seabreeze 0.5.1 to easy-install.pth file
Installing seabreeze-compare script to /usr/local/bin

Installed /home/pi/spec/python-seabreeze
Processing dependencies for seabreeze==0.5.1
Finished processing dependencies for seabreeze==0.5.1
pi@opentrons1 ~/spec/python-seabreeze $ python
Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import seabreeze.spectrometers
>>> import seabreeze.spectrometers as sb
>>> sb.list
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'list'
>>> spec = sb.Spectrometer(sb.list_devices()[0])
>>> spec.serial_number
'USB2K10109'
>>> 
pi@opentrons1 ~/spec/python-seabreeze $
```

ap--, 2015-11-11:
Hi,

your first issue shows up, because you are trying to import seabreeze while your current working directory is the git repository root. Just cd to a different folder after installing and it'll work.

Regarding your second issue: which osx version are you on? My guess is: it's just pip's version naming system that prevents the installation.

ap--, 2016-02-13:
The osx arch issue should be fixed now (0.5.2).

commit 899905a0 (2016-02-16): [travis] use python.org python interpreters

Closes #19

* python.org python interpreters are fat binary builds and support
  building wheels for --arch=intel
* includes building wheels for python 3.5
closed 2016-02-16 by ap--