ap--/python-seabreeze #21: libseabreeze.so needs to install to /usr/lib not /usr/local/lib on rasbian

alisdairrobertson, 2016-02-20:
If libseabreeze.so is installed to /usr/local/lib on raspian jessie then the default installation of python does not pick up the shared object and so you can't use cseabreeze.
I'm sure you could also add /usr/local/lib to one of the dynamic linker variables as indicated here: http://stackoverflow.com/questions/9542357/how-do-so-files-work-and-where-do-they-go-in-unix

Pretty new to python so sorry if I've completely missed something obvious. Manually moving libseabreeze.so to /usr/lib was the easiest way I found to get my pi to see an STS.

ap--, 2016-02-21:
Note: maybe also fix https://sourceforge.net/p/seabreeze/tickets/9/

ap--, 2016-02-21:
Thanks for filing the issue.
I won't make it a default option to install to /usr/lib, so we should explain this in the instructions

commit 4c97b196 (2016-03-19): Rewriting install instructions and documentation

* fixes #21
* fixes #20
* fixes #14
commit 47599a36 (2016-03-19): Rewriting install instructions and documentation

* fixes #21
* fixes #20
* fixes #14
closed 2016-03-19 by ap--