ap--/python-seabreeze #22: Can't detect serial number.

Bisaloo, 2016-03-15:
I am not sure where I should report this bug. Maybe it is juste the seabreeze API being faulty but I can't detect my spectrometer, reportedly because it has no serial number. The spectrometer does work fine with spectrasuite on windows.

It fails with both backends.

I'm on ubuntu 14.04 using python 2.7.

You can see the full trace (in ipython) in the attached files.

[cseabreeze_trace.txt](https://github.com/ap--/python-seabreeze/files/174077/cseabreeze_trace.txt)
[pyseabreeze_trace.txt](https://github.com/ap--/python-seabreeze/files/174076/pyseabreeze_trace.txt)

ap--, 2016-03-15:
Looks like a permission problem.
Try running your code as root. If it works for your root user, then you didn't install the udev rules correctly, or your user is not in the plugdev group.

Bisaloo, 2016-03-15:
Thank you for your answer. It does indeed detect the spectrometer when run as root.

But I am confused, I followed your instructions in the readme and:
 * `getent group plugdev` returns `plugdev:x:46:hugo`
 * `ll /etc/udev/rules.d/10-oceanoptics.rules` return `-rw-rw-r-- 1 root 3,3K sept. 30  2011 /etc/udev/rules.d/10-oceanoptics.rules` and the contents of the file look fine...

ap--, 2016-03-15:
try one of the following:
  * `sudo udevadm control --reload-rules`, physically replug your spectrometer and logout and back in as hugo
  *  reboot your computer

Bisaloo, 2016-03-15:
I tried to reboot and still get the same error.

ap--, 2016-03-15:
```lsusb -d 2457:```

Bisaloo, 2016-03-15:
`lsusb -d 2457:`

> Bus 003 Device 002: ID 2457:1022

ap--, 2016-03-15:
Then after connecting your spectrometer you should see a usb4000-something symlink in /dev/ which should be owned by the plugdev group.
If not, either the udev rules are wrong, or your udev installation has problems. I can't really help with either.

Please note, that there are two versions of udev rules floating around:
* https://github.com/ap--/python-seabreeze/blob/master/misc/10-oceanoptics.rules
* https://github.com/ap--/python-oceanoptics/blob/master/misc/10-oceanoptics.rules

The first one works for me.

Bisaloo, 2016-03-15:
Ok, I totally overlooked the fact the my 10-oceanoptics.rules had `SYSFS{}` instead of `ATTR{}`. I'm not sure where that came from...

Anyway, copying the file in the misc folder (I must have somehow skipped this step during the installation process) and rebooting did fix my issue.

Sorry about that, it looks like the issue was definitely on my end. Thank you for your help!

closed 2016-03-15 by Bisaloo