ap--/python-seabreeze #25: Setting up

matsonjr, 2016-05-31:
Is there something else to do to set this up besides installing SeaBreeze and the given .msi? Do I do something with the files on the main page? Because when I try to test it in Idle or in a python file I get errors

ap--, 2016-05-31:
Here are the [Windows installation instructions](https://github.com/ap--/python-seabreeze/blob/master/docs/WINDOWS_INSTALL.md).

matsonjr, 2016-05-31:
Okay, I've done all that. Everything went smoothly but Idle says there's no module to use, and I can't find one. Is that supposed to be installed by the .whl? I installed it in cmd, but I can't say I noticed that anything changed

matsonjr, 2016-05-31:
Do I need to download the master file and put it somewhere to be able to use?

ap--, 2016-05-31:
Since you are not providing any detailed description of what you are doing and/or the resulting error message, I would only be able to guess what's wrong.

If this is the first time you are using python, please consider reading the [pip documentation](https://pip.pypa.io/en/stable/) and [IDLE documentation](https://docs.python.org/2/library/idle.html).

matsonjr, 2016-05-31:
Oh wow I'm sorry my bad. I'm on a windows 7 system. I followed all the installation instructions for windows. However, trying to import the module on IDLE told me it couldn't find the modules, so I redownloaded the master file from the homepage and tried running idle from there, and instead I got an error saying it couldn't get the backend working. I've used python a bit in the past, but not really in this way, and I'm not sure what the steps are to get it working after installation. Sorry to bother you, I really appreciate your time

ap--, 2016-06-01:
You still haven't told me what you typed in IDLE, and you haven't copied the error message here. I am really sorry, but I currently don't have the time to take you by the hand an walk you through every single command of the installation process. Here are the things I would tell you to do to debug this. Every single step here needs to work:

1.) Install numpy and type `import numpy` in a python interpreter.
2.) Use pip to install the correct python-seabreeze wheel. `import seabreeze` should work now.
3.) Install the seabreeze C-library from the ocean-optics sourceforge site. `import seabreeze.cseabreeze`  should work now.

If you can't get step 1 and 2 working, you really need to read some Python tutorials. The issue is then most likely on your side. If step 3 fails, look at the Windows Installation instructions and do the thing that tells you to add the folder that contains the seabreeze.dll to your %PATH% environment variable.

If you are 100% sure that you did step 3 correctly, but it didn't work, AND you added the Seabreeze.dll folder to your %PATH% and it still didn't work, give a very very very detailed summary of what you were doing (copy your code and error messages) and report back here. I might be able to help you then.

All the best,
-Andreas

ap--, 2017-05-27:
Now that anaconda packages are provided, this should not be an issue anymore.

closed 2017-05-27 by ap--