ap--/python-seabreeze #26: seabreeze 3.0.11

iCagarro, 2016-06-06:
I'm using the new Ocean Optics flameNIR spectrometer and I have seen that this Python module does not support this device yet. However, there is a new version of the official SeaBreeze Embedded Open-Source Device Driver that allows the use of this spectrometer. I have substituted the old .zip file, changing the organization of it, and I have been able to connect and use the flameNIR spectrometer using Python under a Raspberry Pi 2 and 3. It would be nice to update the installation files, so the installation process could be easier for general users.

Thank you for developing this awesome project.

ap--, 2016-06-06:
Hi iCagarro,

the main reason why I didn't update to 3.0.11. yet was, that the [sourceforge site](https://sourceforge.net/projects/seabreeze/files/SeaBreeze/installers/) is missing a 32bit installer for windows.

But I'll see what I can do. This should not be a blocking issue.
So can you confirm that **seabreeze-3.0.11**  and the **cseabreeze backend** work fine with **python-seabreeze-0.5.3** (or current master)? ( maybe I should use a better naming scheme... 😉  )

Thanks a lot,
Andreas

iCagarro, 2016-06-06:
Hello Andreas,

Yes, I can confirm that seabreeze-3.0.11 and the cseabreeze backend work fine with the current master. I'm using two spectrometers at the same time having no troubles.

As I said, thank you for all the work behind this module,
Adrià

ap--, 2016-06-06:
Cool, thanks.
I think I'll update to _3.0.11._ soon then. I'll just drop windows 32bit support for now, until Ocean Optics provides the installers.

ksquire10, 2016-08-12:
I am new to python and I am trying to do the same thing as iCagarro and having a difficult time.  To integrate the new Seabreeze3.0.11 do I need to create a new .pyd file and swap it out for the wrapper.pyd file in the cseabreeze file?  Is there a better way?

ksquire10, 2016-08-17:
Hey Andreas,

I'm new to python and trying to control the FlameNIR.  Is there an easy way I can integrate seabreeze-3.0.11 with python-seabreeze or should I try and figure out how to change the seabreeze files like Adria?

ap--, 2017-05-27:
Now that anaconda packages are provided and ship version libseabreeze 3.0.11+, this should not be an issue anymore.

closed 2017-05-27 by ap--