ap--/python-seabreeze #27: how to use python-seabreeze with anaconda python

zhengqinggan, 2016-06-09:
Operating system (Linux distribution, Windows version, OSX version) and archictecture (32bit, 64bit, arm) : window 7 64bit
Python version and arch (i.e. Python 2.7.10 64bit): Anadonda python 3.5 64bit
python-seabreeze version: seabreeze-0.5.3-cp35-cp35m-win_amd64.whl

when I run import seabreeze.spectrometers, it shows the error as bellow:

warnings.warn("Can't load seabreeze c library wrapper, because:\n'%s'\n" % err.message )
AttributeError: 'ImportError' object has no attribute 'message'

in the same computer, if I run the python-seabreeze with python2.7 32bit, it works fine.

Thank you very much!

ap--, 2016-06-09:
Thanks for reporting!
I already pushed a fix.
The next release version will work as expected. (It'll take a little bit, until i fix #26 and make a new release)

In the meantime you can workaround this error by running:

```
import seabreeze
seabreeze.use('pyseabreeze')
import seabreeze.spectrometers
```

zhengqinggan, 2016-06-10:
thanks for the suggestion, but I still get some errors using the suggested code:
    raise ValueError('No backend available')
ValueError: No backend available

ap--, 2016-06-12:
Do you want to use the **cseabreeze** or **pyseabreeze** backend?

zhengqinggan, 2016-06-14:
I don't have preference, just try to make it work with anaconda 3 python 3.5

ap--, 2017-05-18:
Anaconda is now the recommended install method

closed 2017-05-18 by ap--