ap--/python-seabreeze #28: High speed acquisition mode

MebaR, 2016-07-04:
I am using HR2000+ spectrometers and for my setup I will need the high speed acquisition mode as provided by the OmniDriver. Means: preallocating internal buffer, recording spectra until this buffer is filled and then read out the data from the spectrometer. Is this possible?
Thanks.

ap--, 2016-07-06:
Please contact ocean optics support, and ask them if the seabreeze library supports this.
If the answer is yes, this module supports it.

nharmon9, 2017-02-07:
Any development on this?

ap--, 2020-08-13:
The HR2000+ does not have support for preallocating an internal buffer in USB mode according to the datasheet I have.
The USB Highspeed mode is supported by both python-seabreeze backends.
I'm closing this issue.

Please reopen, if I understood something wrong.

Cheers,
Andreas 😃

closed 2020-08-13 by ap--