ap--/python-seabreeze #29: Issue Recognizing Spark Using Raspberry Pi and Python

bkinard4, 2016-07-25:
Hi.  First, thanks for this software.  I have a Flame (USB2000+) and Spark spectrometer.  They both work properly - confirmed using Ocean Optic's OceanView software for the Mac.  I need an embedded application, however, using the Spark.  I updated all of the software on the Raspberry Pi 3 (Raspian, Python 2.7.10 and PIP 8.1.2).  I installed the Python software on the Raspberry Pi 3, and it works fine with my Flame (USB2000+) spectrometer; but I receive the following error message when I attach the Spark spectrometer to the Raspberry Pi 3 and run the script:
[image]

Using the OceanView software on the Mac, I confirm that the Spark spectrometer is identified and working:
[Spark Notification.pdf](https://github.com/ap--/python-seabreeze/files/382678/Spark.Notification.pdf)

Any assistance to resolve this issue would be greatly appreciated.
Thanks!
Brian

ap--, 2016-07-28:
Please try installing version 3.0.11 of the seabreeze C library with the install_libseabreeze.sh script. To do this, change the second line in the script to:
```
SEABREEZE_FILE="seabreeze-3.0.11.zip"
```
and then run it.

After that, rerun your python test code, and report back.

OTTStreamingVideo, 2016-07-28:
I am working with Brian on this issue.

This morning we modified the install_libseabreeze.sh script as you suggested by changing the second line to:

SEABREEZE_FILE="seabreeze-3.0.11.zip"

When we ran the ./install_libseabreeze.sh command we received the following error message:

# SEABREEZE: using seabreeze-3.0.11.zip
# SEABREEZE: extracting seabreeze-3.0.11.zip
Archive:  seabreeze-3.0.11.zip
# SEABREEZE: building shared library
make: *** No rule to make target 'lib/libseabreeze.so'.  Stop.
# SEABREEZE: installing to /usr/local/lib
install: cannot stat ‘seabreeze-3.0.11/lib/libseabreeze.so’: No such file or directory

We then discovered that there is an additional folder in the hierarchy in seabreeze-3.0.11.zip as compared to "seabreeze-3.0.10.zip" by downloading the two zip files onto a Windows laptop only for the purposes of comparing the two zip files.

seabreeze-3.0.10.zip:

07/28/2016  09:12 AM    <DIR>          .
07/28/2016  09:12 AM    <DIR>          ..
07/28/2016  09:11 AM             3,992 common.mk
07/28/2016  09:11 AM    <DIR>          doc
07/28/2016  09:11 AM    <DIR>          include
07/28/2016  09:11 AM             1,127 LICENSE
07/28/2016  09:11 AM             2,960 Makefile
07/28/2016  09:11 AM    <DIR>          os-support
07/28/2016  09:11 AM            31,043 README.txt
07/28/2016  09:12 AM    <DIR>          sample-code
07/28/2016  09:12 AM    <DIR>          scripts
07/28/2016  09:12 AM    <DIR>          src
07/28/2016  09:12 AM    <DIR>          test
07/28/2016  09:11 AM               826 win.mk

seabreeze-3.0.11.zip

07/28/2016  09:10 AM    <DIR>          .
07/28/2016  09:10 AM    <DIR>          ..
07/28/2016  09:09 AM    <DIR>          daemons
07/28/2016  09:10 AM    <DIR>          SeaBreeze
07/28/2016  09:10 AM    <DIR>          util

On the RPi we ran the following command to copy the contents of the SeaBreeze folder one level up:

cd /home/pi/python-seabreeze/misc/seabreeze-3.0.11/SeaBreeze
cp -r ./* ..

After this this install_libseabreeze.sh script appeared to run properly:

cd /home/pi/python-seabreeze/misc
./install_libseabreeze.sh

last output line:
# SEABREEZE: installing to /usr/local/lib

After the install we were able to communicate successfully with the Flame spectrometer but not the Spark spectrometer using these three lines of Python in a test file in the cd /home/pi/python-seabreeze/misc/seabreeze-3.0.11/test folder:

import seabreeze.spectrometers as sb
devices = sb.list_devices()
print devices

Using idle on the RPI, the traceback was:

Traceback (most recent call last) :
File " /home/pi/python-seabreeze/misc/seabreeze-3 .0.11/test/test000.py", line 1
6, in <module>
devices = sb.list_devices()
File "seabreeze/cseabreeze/wrapper,pyx" , line 84 , in seabreeze.cseabreeze:. wrap
per.device_list_devices ( ./seabreeze/seabreeze/wrapper.c :3335)
serial = device_get_serial_number(tmpdevice)
File " seabreeze/cseabreeze/wrapper.pyx." 1 line 133, in seabreeze.cseabreeze. wra
pper.device_get_serial_number ( ./seabreeze/cseabreeze/wrapper.c :4247)
raise SeaBreezeError( "This should not have happened . Apparently this device
has "
SeaBreezeError: This should not have happened. Apparently this device has 0 seri
al number features. The code expects it to have 1 and only 1. Please file a bug
report including a description of your device.

We copied the new 10-oceanoptics.rules file as follows:

cp /home/pi/python-seabreeze/misc/seabreeze-3.0.11/os-support/linux/10-oceanoptics.rules /home/pi/python-seabreeze/misc/10-oceanoptics.rules

cp /home/pi/python-seabreeze/misc/seabreeze-3.0.11/os-support/linux/10-oceanoptics.rules /etc/udev/rules.d/10-oceanoptics.rules

Since we can communicate with the Flame spectrometer but not the Spark spectrometer, is is possible that the issue may be related to the rules file or driver files?

Thank you for your help.

bkinard4, 2016-07-28:
Hi - Just to add to the prior comment, the RPi's OS (Raspian) recognizes the Spark USB.  The Spark Users Manual (page 44) states that the Spark USB Vendor ID is 0x2457 and Product ID is 0x4200.  You will see below that the Raspian recognizes the USB as Device 22 on Bus 001.  
[image]
Thanks again for your help with this!
Brian

ap--, 2016-07-29:
Thanks for the detailed report.
I think you're right regarding the udev rules.
Try running your python script as root. 
If it works as root, add the following line to `/etc/udev/rules.d/10-oceaoptics.rules`

`ATTR{idVendor}=="2457", ATTR{idProduct}=="4200", SYMLINK+="spark-%n", GROUP="plugdev"`

and run `sudo udevadm control --reload-rules` or reboot.

bkinard4, 2016-07-29:
That works!  Thank you!

closed 2016-07-29 by bkinard4