ap--/python-seabreeze #30: Scan averaging?

hughchurchill, 2016-08-04:
Does python-seabreeze support setting a number of scans to average?  For example, 'spectrumProcessingScansToAverageSet' as described in https://oceanoptics.com/api/seabreeze/class_sea_breeze_a_p_i.html

I didn't see it anywhere but sorry if I missed it.  Thanks!

bkinard4, 2016-08-04:
As I understand it, if your signal-to-noise ratio is greater than one, you can increase the ratio further by averaging. You can perform a simple averaging routine in Python by running a "for loop" n times (where n is the number you want to average) and then average the n arrays together by running another iterative loop adding each pixel (or element in the arrays) and then dividing each array number by n.

jonturner53, 2017-01-08:
I think the point of the original question is that the spectrometer can do scan averaging automatically within the spectrometer. Much more efficient than acquiring 20 spectra and averaging them on the host computer. This is a feature that I also would like to use.

ap--, 2017-03-21:
This should now be supported via cseabreeze.
I'll leave it open since pyseabreeze support is still missing.

benjaminesse, 2017-08-08:
Hi, I'm trying to access this feature for my FLAME and USB2000+ but for both I get a seabreeze error returned. I am running windows anaconda python version 3.5.3, but I get the same from my raspberry pi. 

Using the following code
```python
import seabreeze.spectrometers as sb

devices = sb.list_devices()
spec = sb.Spectrometer(devices[0])

spec.scans_to_average(10)
```
I get the following message
```bash
Traceback (most recent call last):

  File "<ipython-input-10-e273bf1c09f5>", line 1, in <module>
    runfile('C:/Users/mqbpwbe2/Dropbox/untitled0.py', wdir='C:/Users/mqbpwbe2/Dropbox')

  File "C:\Anaconda3\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 699, in runfile
    execfile(filename, namespace)

  File "C:\Anaconda3\lib\site-packages\spyderlib\widgets\externalshell\sitecustomize.py", line 88, in execfile
    exec(compile(open(filename, 'rb').read(), filename, 'exec'), namespace)

  File "C:/Users/mqbpwbe2/Dropbox/untitled0.py", line 13, in <module>
    spec.scans_to_average(10)

  File "C:\Anaconda3\lib\site-packages\seabreeze\spectrometers.py", line 188, in scans_to_average
    lib.spectrum_processing_set_scans_to_average(self._dev, self._fidspp, scans_to_average) # Not implemented in pyseabreeze

  File "seabreeze/cseabreeze/wrapper.pyx", line 642, in seabreeze.cseabreeze.wrapper.spectrum_processing_set_scans_to_average (./seabreeze/cseabreeze/wrapper.c:14669)

SeaBreezeError: b'Error: No such feature on device'
```
As I installed through Anaconda I think I should be using the cseabreeze wrapper not the pyseabreeze one.

Many thanks, Ben

ap--, 2017-08-11:
Hi Ben,

that's not a Bug. The spectrum_processing feature is only available on Spark and STS spectrometers.

Best,
Andreas

benjaminesse, 2017-08-11:
Hi Andreas,

Thanks for your reply, I shall stick to my for loops in that case!

Cheers,

Ben

marsteller, 2021-04-14:
Hi, 
we are using a newer QEPro in our lab, and it's supposed to support scan averaging.
If this is not OceanView faking the on-device averaging, would it be possible to also implement scan averaging for this spectrometer?

Best regards,
Alexander