ap--/python-seabreeze #31: Workaround Fedora 24 for non-existing plugdev group

TobiasNils, 2016-08-10:
As the plugdev group should not exist in Fedora (see [https://bugzilla.redhat.com/show_bug.cgi?id=859244](url)), the udev rules provided in "10-oceanoptics.rules" have to be adjusted to give users without root permission access to the spectrometers:

**Replace** `SUBSYSTEM=="usb_device",` in the header with
`SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", MODE="0664"` ,
according to the default _50-udev-default.rules_ in _/lib/udev/rules.d_ and **delete** all the `GROUP="plugdev"`. 
  
I attach a modified 10-oceanoptics.rules and install_udev_rules.sh which work with my recent Fedora 24 installation 

[fedora udev rules.tar.gz](https://github.com/ap--/python-seabreeze/files/411597/fedora.udev.rules.tar.gz)

Stoccarda, 2018-10-19:
Many thanks for publishing the workaround, I should have seen it earlier ...
I tried to install seabreeze under 64-bit Fedora 28.
I additionally had the problem that the recommended conda installation caused a no more working dnf & dnfdragora.
If it is not too much work, what will be the steps for a complete Seabreeze installation under Fedora?

TobiasNils, 2018-12-15:
Sorry for the delay, I just saw your comment. May be a bit late already, but here's what I do to not interfere with the system's Python installation at all: 
1. Install [Anaconda](https://www.continuum.io/downloads) as suggested in [THE RECOMMENDED WAY](https://github.com/ap--/python-seabreeze#installation).
    - Run the Anaconda installation script.
    - The Anaconda installation script asks at the end if want to prepend the conda path your system's PATH variable; *Don't*. 
    - Instead, [create aliases]() for the anaconda binaries to use them from the system shell, e.g. add one or more of the following lines to the end of the `~/.bashrc` file: 
 ```alias conda=/path/to/anaconda3/bin/conda``` for the `conda ` command, 
 ```alias conda=/path/to/anaconda3/bin/spyder``` to launch the [Spyder IDE](https://www.spyder-ide.org/) shipped with Anaconda, 
 ```alias conda=/path/to/anaconda3/bin/jupyter-notebook``` for the [jupyter notebook server](https://jupyter.org/), etc.
2. With Anaconda installed and the alias in place; install python-seabreeze as indicated in the [original post](https://github.com/ap--/python-seabreeze#installation):
  ```  conda install -c poehlmann python-seabreeze  ```
3. Install udev rules as indicated [here](https://github.com/TobiasNils/python-seabreeze#installation) for *Fedora*.  
That should do the trick.

Stoccarda, 2018-12-16:
Thanks for coming back !!!
I will follow your instructions on the PC with Fedora 29 where it will be used with the spectrometer.
On the notebook I am using, anaconda 29.44.7 is already installed with Fedora 29.
But conda and Spyder are not installed and on the anaconda website 
https://www.anaconda.com/download/#linux
version 5.3.1 is the current one .
A bit confusing ...

TobiasNils, 2018-12-17:
Welcome! 
This anaconda 29.44.7 you found already installed is the [Fedora system installer](https://fedoraproject.org/wiki/Anaconda), which is not at all what you are looking for here. 
Just go ahead with [this](https://www.anaconda.com/download/#linux) and run the installer script. It will ask you for an installation directory. *Note the path to this directory or specify one yourself* and use it when setting up the aliases. With that you should be all set.

ap--, 2019-09-11:
With version 1.0.0 the udev rules set MODE 0666 and don't use groups anymore which means it should work on any distribution and fix this bug :)

also there's an installer script that comes with the new version. After installing seabreeze you can run
```
seabreeze_os_setup
```
and it takes care of installing the udev rules for you :tada:

closed 2019-09-11 by ap--