ap--/python-seabreeze #32: STS Sensors and multiprocessing

Fi156, 2016-09-15:


ap--, 2016-09-15:
Please create a minimal example that reproduces the error.

It looks like you are opening the Spectrometers in the main process and then try to retrieve the intensities from different subprocesses. My guess is that if you move the instantiation of the spectrometer objects to the individual subprocesses it might work. But no guarantees. I am not sure if pyUSB is the limiting factor here.

Fi156, 2016-09-15:
Thank you for your quick response!

I have tried to reduce the example to a minimum. But still multiprocessing and sensor initialization are adding some necessary lines of code here - sorry!

http://pastebin.com/XYC3gwDb

Yes, I am opening the Spectrometers and loading these into a list (Line 42).

I tried to put the whole initialisation code into the subprocess part, afterwards even longer sleep commands are needed, because otherwise I get an error "USB device busy".

>  move the instantiation of the spectrometer objects to the individual subprocesses
Or which part should I move into the subprocess?

ap--, 2020-08-13:
Hey there,
I'm cleaning up a few older issues. Sorry I never got back to you on this one 🙇  😞 

- Seabreeze is not multiprocessing safe nor even thread safe.
- Spectrometers should be initialised in their own processes.
- I would not try to pass spectrometer instances via pickling to other processes.
  In its current state, that is basically calling for a disaster to happen 😅 
- I would double check that class decorators don't interfere with the pyseabreeze metaclass usage.

All those comments should have been here 4 years ago. Sorry again 🙇 

Cheers,
Andreas

closed 2020-08-13 by ap--