ap--/python-seabreeze #33: Compatibility with new Flame series

jgrizou, 2016-10-04:
We are thinking of buying the latest Flame series: http://oceanoptics.com/product/flame-spectrometer/

I do not know much about the low-level behind this library. Would the code be compatible?  ([Operating Manual](http://oceanoptics.com/wp-content/uploads/FlameIO.pdf))

If not, is the communication layer compatible, i.e. would it be 'easy' to integrate this device for a regular programmer?

Thanks for your help

ap--, 2016-10-04:
Hi jgrizou,

I recommend you ask ocean optics support, but I think this controls exactly like the USB2000Plus. They even have the same USB Product ID: 0x101E.
So if the above is true, the flame spectrometer should be supported by both the seabreeze library, and the pyseabreeze backend. Therefore it should be supported by python-seabreeze.

And, yes, the communication layer looks compatible. The rest depends on your definition of a _regular programmer_.

ThomasGerstenberg, 2017-06-07:
Thought I'd leave an update since this issue is still marked open:

Just hooked up a Flame-S and was able to use this library out of the box using the `cseabreeze` backend. It is recognized as a USB2000+ from the device driver standpoint and operates the same

ap--, 2017-06-08:
Awesome. Thanks for the report!

closed 2017-06-08 by ap--