ap--/python-seabreeze #34: Integer intensity array instead of float

zoomx, 2016-10-06:
Is there a way to obtain the spectrum as an array of integers instead of floats?

I am not sure that I can obtain it just changing spectrometers.py line 148 from
`lib.spectrometer_get_formatted_spectrum`
to
`lib.spectrometer_get_unformatted_spectrum`

ap--, 2016-10-08:
Hi zoomx,

I am not sure what you want to do, but if you just want to convert the numpy array to integer, you can use `spec. intensities().astype(np.int64)` (you'll loose precision though...).
If you want the raw pixel values, you have to decode the raw encoded data yourself.
`lib.spectrometer_get_unformatted_spectrum` returns the byte array that is encoded in a specific way for each spectrometer. You can look up how to decode for your spectrometer model, by either reading the datasheet for your spectrometer, or looking at the __pyseabreeze__ implementation for your model.

zoomx, 2016-10-10:
Hi ap,
thanks for your reply.
I wrote a program in VB.NET using Omnidriver and I get a spectra using the functions getSpectrum. I get a  array of double (but inside there are 16 bit unsigned integers), 1023 elements, they are the raw values of the CCD.
I use raw values because these are the values requested to me (it is used on another program to evaluate gas concentration written in IDL not by me)  and because there is a sort of autoexposition that works with raw values, it keeps the interesting part of the spectra above 50000 and under 65000. In this way I avoid clipping and underexposition, because acquisition is unattended.
The goal is to build unattended measuring station using a RaspberryPi or another similar board.
So I need only the raw values, 1023 unsigned int values that came from CCD.

ap--, 2016-10-10:
Hi zoomx,
if you call `spec.intensities()` it should do exactly what you want.
If not: please provide an example script, and which spectrometer model you are using, and the output of the call that's supposedly wrong, and a detailed description of what you'd expect it to return.

zoomx, 2016-10-11:
Unfortunately this
```
spectra = spec.intensities()
print (spectra)
```
give this

 [  1.69905556e+01   2.72164428e+04   1.59711222e+03 ...,   1.79371722e+03
   1.79371722e+03   1.79371722e+03]

The function intesities() uses spectrometerGetFormattedSpectrum

Here
http://oceanoptics.com/api/seabreeze/class_sea_breeze_a_p_i.html
is written that 
`spectrometerGetFormattedSpectrum (long deviceID, long spectrometerFeatureID, int *errorCode, double *buffer, int bufferLength)`
so it returns a double array

In spectrometer.py I found

```
    def get_formatted_spectrum(self, out):
        tmp = numpy.empty((self._RAW_SPECTRUM_LEN), dtype=numpy.uint8)
        self.get_unformatted_spectrum(tmp)
        ret = numpy.array(struct.unpack("<" + "H"*self._PIXELS, tmp[:-1]), dtype=numpy.double)
        out[:] = ret * self._NORMALIZATION_VALUE
        return self._PIXELS  # compatibility
        
    def get_unformatted_spectrum(self, out):
        self.usb_send(struct.pack('<B', 0x09))
        out[:] = self.usb_read_highspeed(size=self._RAW_SPECTRUM_LEN,
                        timeout=int(self._INTEGRATION_TIME_MAX * 1e-3 + self.usbtimeout_ms))
        return self._RAW_SPECTRUM_LEN  # compatibility
```

Maybe I am making some mistakes somewhere because I am very new with python but it seems that the unformatted spectrum is the same as formatted and the difference is only that chars was grouped into doubles. I didn't find any other function that retrieve raw spectrum. Maybe I have to ask to OceanOptics.

I am using an USB2000+

ap--, 2016-10-11:
Okay, I think I understand the problem. So let me try to explain it better: 

The seabreeze library rescales the USB2000+ spectra to 65535 using a saturation value that's stored in the eeprom of the spectrometer. My guess is that they do it to cope with variability between the CCD sensors. That's the reason why the intensities() call returns non-integral numbers.

If you want the raw sensor values, you need to either get the data from `get_unformatted_spectrum` and do the conversion to integers yourself (the first three lines of `get_formatted_spectrum`).

Or you could just undo the multiplication with `_NORMALIZATION_VALUE` ([reference](https://github.com/ap--/python-seabreeze/blob/master/seabreeze/pyseabreeze/interfaces/spectrometer.py#L165-L177))

The following is not tested:
```
spec  # <- this is an instance of your spectrometer
# UNTESTED, SHOULD ONLY WORK FOR USB2000+
# _MAX_PIXEL_VALUE = 65535
import struct  
ret = spec.eeprom_read_slot(17) # <- this might not work ...
saturation = struct.unpack("<H", ret[6:8])[0]
revert_normalization = saturation / 65535.0
my_raw_intensities = np.array(spec.intensities() * revert_normalization, dtype=np.int)
print my_raw_intensities
```

If this doesn't work:
1. fully saturate your spectrometer and read the return values
2. Use the maximum return value to rescale the output of `spec.intensities()` to whatever you need it to be and convert it to whatever datatype you require.

But be aware:
If the code that you're sending these spectrums to relies on fixed thresholds, you should probably rewrite it, because right now it will depend on which USB2000+ spectrometer you are using, since the CCD sensor in every different USB2000+ saturates at different values!

zoomx, 2016-10-11:
Thank you very much!

closed 2016-10-11 by zoomx