ap--/python-seabreeze #37: Read temperature feature

iCagarro, 2017-02-22:
Hello,

Is it possible to read the temperature sensor using this library? In the Version History of the Seabreeze API documentation there is a brief note saying: "added read temperature feature" (version 3.0.8).

https://oceanoptics.com/api/seabreeze/

Thank you for your time, this library is awesome.

ap--, 2017-02-23:
Currently the TemperatureFeature is not wrapped in the cseabreeze backend.
And the only two Spectrometers that support it are the STS and the Spark.

But using the pyseabreeze backend it should be possible to jerry-rig it...
**This is untested and _(if it works)_ will only work with a STS or a Spark spectrometer:**
```python
import seabreeze
seabreeze.use('pyseabreeze')
import seabreeze.spectrometers as sb
import struct

# open first spectrometer
spec = sb.Spectrometer.from_serial_number()
# query number of temperature sensors
_N = spec._dev.interface.query(0x00400000, "")
N = struct.unpack("<B", _N)[0]
print "N =", N
# read temperatures
for i in range(N):
    try:
        temp = spec._dev.interface.query(0x00400001, struct.pack("<B", i))
    except Exception as e:
        print "Error reading sensor", i, "('%s')" % str(e)
    else:
        print "temp", i, "=", struct.unpack("<f", temp)
```

iCagarro, 2017-03-06:
I have tested it with a STS and the output is:

N = 3
temp 0 = (29.172489166259766,)
temp 1 = (-34.25904846191406,)
temp 2 = (55.93994903564453,)

I think that temp 0 value corresponds to the temperature in Celsius.

ap--, 2017-03-06:
From the [STS datasheet page 30](https://oceanoptics.com//wp-content/uploads/STS-Data-Sheet.pdf):

> The microcontroller sensor will report values much higher than the detector board
> thermistor because the microcontroller integrated circuit runs at a higher temperature.
> The STS contains three memory locations for the temperature sensor as follows:
>    0 = Detector Board Thermistor
>    1 = Reserved/Internal Use
>    2 = Microcontroller Sensor Temperature 

And:
> Units are in °C.

ap--, 2019-09-11:
with version 1.0.0 all supported SeaBreeze features are wrapped, which means, it should be possible to do this now :tada: 

```python
# if your spectometer supports the temperature feature
from seabreeze.spectrometers import Spectrometer
spec = Spectrometer.from_first_available()
spec.f.temperature.temperature_get_all()
```
see:
https://python-seabreeze.readthedocs.io/en/latest/backend_api.html#seabreeze.cseabreeze.SeaBreezeTemperatureFeature

closed 2019-09-11 by ap--