ap--/python-seabreeze #38: Same result with different integration time

XiaomingY, 2017-03-13:
Hi, I am new to spectrometer and Seabreeze. I am trying to use STS spectrometer on a raspberry Pi for daylight studies. I followed the documentation and example code. The intensities returned always has similar values does not matter what integration time I set or whether sensor is covered or not.  Output is saved [here](https://fosterandpartners-my.sharepoint.com/personal/xiyang_fosterandpartners_com/_layouts/15/guestaccess.aspx?docid=09caab819f2ea4775b3795725049ac411&authkey=AQRM3xtTYF7Pr5wdB-ImdFQ)

####################################
import seabreeze.spectrometers as sb
import scipy.io as sio
import time
import numpy as np

devices = sb.list_devices()
spec = sb.Spectrometer(devices[0])

spec.trigger_mode(0)
spec.integration_time_micros(100)
spec.wavelengths()
WaveLengths = spec.wavelengths()
Intensities = spec.intensities(correct_dark_counts=False, correct_nonlinearity=True)

f_handle = file('Output.txt', 'w')
for index, val in enumerate(Intensities):
    f_handle.write(str(WaveLengths[index])+" "+str(Intensities[index])+"\n")
f_handle.close()
#######################################

ap--, 2017-03-13:
What happens if you use an integration time of one second instead of 100 micro-seconds?

XiaomingY, 2017-03-14:
Thank you for your reply. I did try different integration time, 1000, 2000... I also cover and uncover the sensor head. The result is not the same but always very similar result. I plotted the result, the shape of the spectrum is very similar and the range of the values is very similar.

ap--, 2017-03-15:
try setting your trigger mode to free running and take several spectra, if that wont work, report to oceanoptics.

```Python
spec.trigger_mode(2)
spec.integration_time_micros(100000)
for i in range(10):
    intensities = spec.intensities(correct_dark_counts=False, correct_nonlinearity=True)
print intensities
```

XiaomingY, 2017-03-15:
Thank you for the suggestions. I changed the trigger mode to 2 then I have the following error messages:
Traceback (most recent call last):
  File "saveSpectrum2File_.py", line 18, in <module>
  Intensities = spec.intensities(correct_dark_counts=False, correct_nonlinearity=True)
  File "/usr/local/lib/python2.7/dist-packages/seabreeze/spectrometers.py", line 149, in intensities
  out[transfered_N:])
  File "seabreeze/cseabreeze/wrapper.pyx", line 209, in seabreeze.cseabreeze.wrapper.spectrometer_get_formatted_spectrum (./seabreeze/cseabreeze/wrapper.c:5729)
seabreeze.cseabreeze.wrapper.SeaBreezeError: Error: Data transfer error

Normally I connect the spectrometer to USB,  a led light on the spectrometer is on. When I use trigger_mode(2) or trigger_mode(1), the script freeze and the led light is off. If I unplug the USB cable from the spectrometer, I have the above error messages.
Any suggestions on this? 

I also tried trigger_mode(0) with a loop to get the last reading from a series of readings. If I cap the sensor head, the dark reading is very high (compared to the dark reading I got from OceanView, 1500 compared to 50). If I remove the cap, the reading is reasonable. Very strange behavior.

ap--, 2017-03-16:
It's very hard to debug from here. I would recommend you'll try the following things:
 - get the python module to work on the same machine you run OceanView
 - try disabling the nonlinearity correction
 - try not setting the trigger mode at all (then it should default, to probably 0 in your case)
 - try both backends. cseabreeze and pyseabreeze.

Good luck :)

XiaomingY, 2017-03-20:
Thank you very much for your suggestions. I will update my findings if you have any progress.

ap--, 2017-06-08:
Closing this for now. Please reopen if this is still an issue.

closed 2017-06-08 by ap--

lancaster17, 2017-11-21:
I am experiencing similar issues. If I get and plot the spectrum with 1 s acquisition time, then change the time to 10 s and measure again (like shown below) i am going to get the result immediately (not blocking for 10 s) and the result would look exactly as for 1 s. If i repeat the measurement again I will be getting proper 10 s measurement.:
```python
spec.integration_time_micros(1e6)  
plot(spec.intensities())
spec.integration_time_micros(1e7)
plot(spec.intensities())
```
I believe this is due to the fact that the spectrometer is in free-running mode and returns the previous measurement when the 10 s measurement is still running. It does not work for me, as I need to measure spectra in a loop and make sure that the acquisition starts exactly at the beginning each iteration and not before. Therefore I tried to do `spec.trigger_mode(1)`, which according to seabreeze docs it the "software-controlled" trigger mode, but if i try to measure in this mode the code freezes exactly as stated above.  I wonder if it is waiting to receive a trigger signal and if there is a specific way to send it.

I tried both backends and do not see any difference in behaviour.
Also I am experiencing similar dark count values of approximately 1500 counts, however I did not check what are the dark counts values in OceanView. I am not using dark count or nonlinearity corrections. And  I checked that `spec.trigger_mode(0)` is actually the default one.

I will be very grateful if you could help me with this,

Thanks,
Anton

ap--, 2017-11-21:
Hi Anton, you forgot to mention your spectrometer model. Best, Andreas

lancaster17, 2017-11-21:
Hi Andreas,
Yes i did, and thanks to you pointing this out I actually looked trough the spectrometer datasheet. The spectrometer model is QEPro and it seems (according to the datasheet [https://oceanoptics.com/wp-content/uploads/OEM-Data-Sheet-QEPRO.pdf](url)) that it actually does not have the software-controlled trigger mode, so free-running is all I get. It seems that the only way to achieve the desired behaviour is to:

1. Send **Abort Acquisition** message to put the spectrometer into Idle Mode
2. Set it to free running mode at the beginning if each iteration with `spec.trigger_mode(0)`
3. Clear all buffered spectra (Perhaps with the implementation of `sbapi_data_buffer_clear` method)
4. Take the spectrum
5. Send **Abort Acquisition** message again

Do you agree with this findings? And are the "Abort Acquisition" and "Clear buffer" methods implemented in the python library?

Thanks,
Anton

ap--, 2017-11-23:
Hi Anton, I think it's worth a try.

The commands are not exposed via the wrapper, but you can implement them quite easily if you use the pyseabreeze backend.

If you look at issue #37 you'll see an example of how to send the obp commands to the spectrometer.

Let me know how things go, and if you need further help, open a new issue for this.

Best
Andreas

lancaster17, 2017-11-23:
Hi Andreas,
I actually already wrapped the libseabreeze 'sbapi_data_buffer_clear' method in similar manner you did for others and it seems to work. Also I found that the fork mentioned in pull request has the 'spec.write_usb' method already implemented and it also works. I may look into pyseabreze backend at some point, thanks for the example, but now I seem to have everything I need.

You may also be interested in knowing that this was built on Windows 10, with the latest official libraries from sourceforge.

Best,
Anton

ap--, 2017-11-24:
Cool! 
Feel free to make a pull request.
When I have more time I'll start merging new features.

PyMarc2, 2021-10-29:
hello! I'm experiencing the same issues as @lancaster17. Did you push your update? If not how could I get this working? I'm a hardware guy. Thank you very much!!

zzwyrmslayer, 2021-12-06:
I am using c++ wrapper classes to control QE pro. And I just met same problem.I find a way to solve it.
1.set the buffer to minimu size 1.
{
	long test = seabreeze_get_buffer_capacity_minimum(m_iSpectralmeterHandle, &error);
	seabreeze_set_buffer_capacity(m_iSpectralmeterHandle, &error, test);
}
2.call function to clear buffer before readout data.
{
                seabreeze_clear_buffer(m_iSpectralmeterHandle, &error);
		flag = seabreeze_get_formatted_spectrum(m_iSpectralmeterHandle, &error, spectrum, spec_length);
}

PyMarc2, 2022-01-27:
@XiaomingY Can you reopen this issue ?

PyMarc2, 2022-01-27:
> Hi @PyMarc2 <https://github.com/PyMarc2>
> I recommend opening a new issue for your specific problem.
> It's best to include as much information as possible to make it easy to help.
> Example code and stack traces of the error are extremely helpful as well as information regarding you spectrometermodel, your operating system etc.
>
> Cheers,
> Andreas 😃

Thanks for the fast response!
Will do!

Marc :)

positron96, 2024-04-16:
Hello, I am having the same problem with QE Pro spectrometer as I want to start measurement at exact time, controlled by software.

I've read in this issue that QE Pro doesn't have a software trigger, however, the User manual explicitly states that software trigger is supported: https://www.oceaninsight.com/globalassets/catalog-blocks-and-images/manual--instruction-re-branded/spectrometers/ocean-qe-pro_manual.pdf (p 15). Is the manual wrong on this? Does anyone know is they've added this functionality since?

Anyway, it seems some people have solution to this problem, @lancaster17, @zzwyrmslayer , could you say whether you achieved the result with pyseabreeze or cseabreeze?

By the way, link to datasheet some posts ago is dead, and I cannot find any updated datasheet on the official website.

positron96, 2024-05-01:
For future reference, here is my solution for adding "stop acquisition" and "clear buffers" functionality. There were some hacky steps needed because of library architecture. 

https://gist.github.com/positron96/42107b3a8d5d1aa4a92514665cc46dbb