ap--/python-seabreeze #39: Odd error on macOS Python 3.5, setting integration time fails

JuanCab, 2017-03-14:
I had successfully used this to access my spectrometer immediately after installing seabreeze, but now, a few hours later, I am having problems.  Here is a capture of the iPython session:

```
In [1]: import matplotlib.pyplot as plt
   ...: import seabreeze.spectrometers as sb
   ...:

In [2]: devices = sb.list_devices()
   ...: spec = sb.Spectrometer(devices[0])
   ...:

In [3]: print("Attaching ", spec.model, " serial number ", spec.serial_number)
Attaching  USB2000  serial number  USB2G6353

In [4]: spec.integration_time_micros(1000)
---------------------------------------------------------------------------
SeaBreezeError                            Traceback (most recent call last)
<ipython-input-4-28900482d9af> in <module>()
----> 1 spec.integration_time_micros(1000)

/Users/juan/anaconda3/lib/python3.5/site-packages/seabreeze/spectrometers.py in integration_time_micros(self, integration_time_micros)
    166     def integration_time_micros(self, integration_time_micros):
    167         lib.spectrometer_set_integration_time_micros(self._dev,
--> 168                                                      self._fidsp, integration_time_micros)
    169
    170     def trigger_mode(self, mode):

seabreeze/cseabreeze/wrapper.pyx in seabreeze.cseabreeze.wrapper.spectrometer_set_integration_time_micros (./seabreeze/cseabreeze/wrapper.c:5102)()

SeaBreezeError: b'Error: Undefined error'
```

Any ideas why setting the integration time is failing?

ap--, 2017-03-15:
Hmmm.
This is raised by the seabreeze C library...
Can you wrap the integration_time call like this, and report back which error code it returns:
```Python
try:
    spec.integration_time_micros(1000)
except Exception as e:
    print("Seabreeze returns ERROR_CODE: %d" % e.error_code)
    raise
```
But I am pretty sure you will need to report this to the [seabreeze c library repository](https://sourceforge.net/projects/seabreeze/)

JuanCab, 2017-03-15:
Seabreeze returns ERROR_CODE: 1

ap--, 2017-03-15:
[SeaBreezeAPI::spectrometerSetIntegrationTimeMicros](https://sourceforge.net/p/seabreeze/code/HEAD/tree/releases/SeaBreeze/3.0.11/SeaBreeze/src/api/seabreezeapi/SeaBreezeAPI.cpp#l533) should not return 1.

ap--, 2017-03-15:
Just an idea.
Maybe when it was working, you were actually using the pyseabreeze backend instead of the cseabreeze backend.

try:
```Python
import seabreeze
seabreeze.use('pyseabreeze')
import seabreeze.spectrometers as sb
```
to force using the pyusb backend

JuanCab, 2017-03-15:
I was using cseabreeze, not the pyseabreeze interface.  I don't have pyusb installed.

JuanCab, 2017-03-17:
I went through install pyusb (under conda) in Python 2.7, but when I call
```
from __future__ import print_function, division
import matplotlib.pyplot as plt
import seabreeze
# The following line triggers the pyusb interface
seabreeze.use('pyseabreeze')
import seabreeze.spectrometers as sb
```

I get

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-3-8081e0f253ea> in <module>()
----> 1 import seabreeze.spectrometers as sb

/Users/xxx/anaconda3/envs/py27/lib/python2.7/site-packages/seabreeze/spectrometers.py in <module>()
      2 import seabreeze.backends
      3 # get the backend and add some functions/classes to this module
----> 4 lib = seabreeze.backends.get_backend()
      5
      6 list_devices = lib.device_list_devices

/Users/xxx/anaconda3/envs/py27/lib/python2.7/site-packages/seabreeze/backends.pyc in get_backend()
     19         if b: return b
     20     if __SEABREEZE_BACKEND == 'pyseabreeze':
---> 21         b = _use_pyseabreeze()
     22         if b: return b
     23         warnings.warn("Falling back to 'cseabreeze'.")

/Users/xxx/anaconda3/envs/py27/lib/python2.7/site-packages/seabreeze/backends.pyc in _use_pyseabreeze()
     43         return None
     44     else:
---> 45         sbb.initialize()
     46         return sbb
     47

/Users/xxx/anaconda3/envs/py27/lib/python2.7/site-packages/seabreeze/pyseabreeze/wrapper.pyc in initialize()
     29     match = lambda dev: (dev.idVendor == defines.VendorId and
     30                          dev.idProduct in defines.ProductIds)
---> 31     devices = usb.core.find(find_all=True, custom_match=match)
     32     # create the handles and associate protocols
     33     for dev in devices:

/Users/xxx/anaconda3/envs/py27/lib/python2.7/site-packages/usb/core.pyc in find(find_all, backend, custom_match, **args)
   1197                 break
   1198         else:
-> 1199             raise ValueError('No backend available')
   1200
   1201     k, v = args.keys(), args.values()

ValueError: No backend available
```

ap--, 2017-03-17:
if
```Python
import usb.core
list(usb.core.find(find_all=True))
```
fails you did not install pyusb correctly.

Since you're on OSX:
```bash
conda create -n myenv
source activate myenv
conda install -c akode pyusb
python -c "import usb.core; print (list(usb.core.find(find_all=True)))"
```

JuanCab, 2017-03-17:
Well, I don't know why, but did actually use
`conda install -c akode pyusb`
in my Python 2.7 environment, and it claims a successful install, but
`python -c "import usb.core; print (list(usb.core.find(find_all=True)))"`
fails.  

In any case, I was originally using the cseabreeze backend overtime, so while pyusb isn't installing properly, it is also not the fundamental problem.  For some reason SeaBreezeAPI::spectrometerSetIntegrationTimeMicros returned a 1.  I reported it on SourceForge, we'll see how that goes.

ap--, 2017-03-17:
I have libusb and libusb-compat installed via homebrew:
```bash
brew install libusb libusb-compat
```
And my pyusb is using libusb as the default backend.

JuanCab, 2017-03-17:
OK, that got the pyusb library working.  I should have realized it assumed other libraries were necessary.  unfortunately, I got an error, but examining it:

`SeaBreezeError: Integration time must be in range(3000, 655350000).
`

And that was it, my integration time was too short!  I don't know why the error was not caught before, but it might be worth a check in spec.integration_time_micros() to check that the time is allowed.

ap--, 2017-03-17:
Hmm. This check should happen in cseabreeze.
SeaBreezeAPI::spectrometerSetIntegrationTimeMicros should return errorcode 8.

But glad it works now.

_NOTE TO SELF: protect [set_integration_time](https://github.com/ap--/python-seabreeze/blob/master/seabreeze/spectrometers.py#L166): self._minimum_integration_time_micros <= input_

commit bf9876bb (2017-05-18): Adding workaround for oob integration time bug in libseabreeze.

Closes #39
closed 2017-05-18 by ap--