ap--/python-seabreeze #40: non-linearity corrections lead to infinities

JuanCab, 2017-03-17:
Running

`intens = spec.intensities(correct_dark_counts=True)`

returns dark corrected counts, but running

`intens = spec.intensities(correct_dark_counts=True, correct_nonlinearity=True)`

returns infinities, as if the correction involved dividing by zero.  Is there a possibility that to get nonlinearity corrections I have to do some particular exposures first?  Sorry, looked in the limited documentation and didn't see anything obvious.

ap--, 2017-03-19:
The non-linearity correction uses coefficients stored on the spectrometer.
They are stored in the Spectrometer class in spec._nc
Can you report the contents of spec._nc?
Maybe we need to protect against nonsensical nonlinearity coefficients in pyseabreeze.

JuanCab, 2017-03-20:
I tried it and got a back a zero...

```
In [1]: import seabreeze.spectrometers as sb
In [2]: devices = sb.list_devices()
   ...: spec = sb.Spectrometer(devices[0])
   ...:
In [3]: spec._nc
Out[3]: poly1d([ 0.])
```

So it looks non-sensical if you divide by this.  Is this something I should contact Ocean Optics about?  Maybe they have the calibration information saved?

ap--, 2017-03-20:
So it could be, that your spectrometer just doesn't have any coefficients stored or it could be a problem with pyseabreeze. Report back with the output of:
```Python
for i in range(6, 14):
    print "'%s'" % spec.eeprom_read_slot(self, i)
```

ap--, 2017-03-21:
I think this is a bug in pyseabreeze. The order returned by your USB2000 spec.eeprom_read_slot(14) will be 0.

In this case [cseabreeze](https://sourceforge.net/p/seabreeze/code/HEAD/tree/trunk/SeaBreeze/src/api/seabreezeapi/NonlinearityCoeffsFeatureAdapter.cpp#l66) will internally return a zero length vector and consequently raise an Exception, that the device does not support nonlinearity correction.

[Pyseabreeze](https://github.com/ap--/python-seabreeze/blob/master/seabreeze/pyseabreeze/interfaces/nonlinearity.py#L16-L22), currently just assumes that there are 8 coefficients stored, and will return a list with eight zeros.

JuanCab, 2017-03-21:
Yep, it returns all zeros:
```
order: 'b'0\x00\x00\x000000e+000\x00\x00''
'b'0.000000e+000\x00\x00''
'b'0.000000e+000\x00\x00''
'b'0.000000e+000\x00\x00''
'b'0.000000e+000\x00\x00''
'b'0.000000e+000\x00\x00''
'b'0.000000e+000\x00\x00''
'b'0.000000e+000\x00\x00''
'b'0.000000e+000\x00\x00''
```

So I guess the USB2000 doesn't support nonlinearity corrections.

ap--, 2017-03-21:
> So I guess the USB2000 doesn't support nonlinearity corrections.

Well... to be nitpicky: you could still do nonlinearity correction, the device just doesn't store predetermined coefficients.

In a way if you are concerned about the linearity of your relative intensity measures, you should measure the nonlinearity coefficients PER PIXEL and not use the predetermined 7th order polynomial for the whole sensor that's (maybe) stored on the device.

But back to topic: pyseabreeze should raise an error when the polynomial order is 0.

closed 2017-03-21 by ap--