ap--/python-seabreeze #44: Can't load c library wrapper

Ecollot, 2017-04-18:
Hi,

I followed all the steps to install cseabreeze (and checked at the end that WinUSB was added to my spectrometer name in device manager).
The wheel corresponding to my python version (i.e win64, python 2.7) would not install however, I had to install the one for win32. But the install seemed to work fine.

I installed seabreeze from an .msi file, so did not go through the steps for building Seabreeze, as specified in http://oceanoptics.com/api/seabreeze/

When running the command in the iPython console "import seabreeze.spectrometers as sb", I get the following error:
[image].

I tried running the "CSharpDemo.csproj", but nothing happens when I double-click on it.

I also tried running the "api_test.c" in Sublime Text 2 and got the following error:
[image]

Any idea what I'm doing wrong?

Best regards,
Teratora

ap--, 2017-04-19:
I think you are running a 32bit python interpreter and you installed the 64bit seabreeze library.

What does ```python --version``` return?
And please post the output of

```Python
import sys
print "%x" % sys.maxsize
```

ap--, 2017-06-08:
Closing this for now. Please reopen if this is still an issue.

closed 2017-06-08 by ap--