ap--/python-seabreeze #46: Anaconda and Raspberry Pi

Pramuspl, 2017-05-22:
I tried to install pyseabreeze using Miniconda (full Anaconda is not available for Raspberry Pi) but I can never get past `conda install -c poehlmann python-seabreeze` - it says theres no such package

ap--, 2017-05-27:
Hi Pramuspl,

I now made prebuilt packages for Raspberry PI 2&3 for conda (armv7l).

And just now I saw, that you have a Raspberry Pi 1 (armv6l)...
If you need me to build armv6l packages, donate 44.99Eur so that I can order a PI B+ kit and ship armv6l in the future too.

I'll leave this open in case someone else needs armv6l too.

Pramuspl, 2017-06-02:
I went the hard (not recommended) way and it worked.

alantate, 2017-06-20:
'conda install -c poehlmann python-seabreeze' does not work. The package is located in the User Channel. Instead, try:

conda install --channel https://conda.anaconda.org/poehlmann python-seabreeze

This worked for me (June 20, 2017) on a Beaglebone Black Wireless running Debian Jessie (which is similar to Raspbian on the Raspberry Pi 3B), running Berryconda3 and conda 4.3.22-py35_0 (the current update).

Nice work on this Mr. Poehlmann.

ap--, 2017-06-21:
@alantate thanks!

vxgmichel, 2017-08-07:
@ap-- I got a raspberry pie 1 and I'd like to try building a conda package for it. Could you share the recipe you used for `armv7l`?

ap--, 2017-08-07:
@vxgmichel here you are: https://github.com/ap--/python-seabreeze-feedstock/tree/master/recipe

vxgmichel, 2017-08-07:
@ap-- Thanks! Also I just remembered the recipe is actually included in the package. 

By the way, I noticed two differences between the recipe from the repo and the one from the package:
```diff
diff --git a/recipe/build.sh b/recipe/build.sh
index 8ca252b..715e3b1 100644
--- a/recipe/build.sh
+++ b/recipe/build.sh
@@ -6,7 +6,7 @@ if [ "$(uname)" == "Darwin" ]; then
 elif [ "$(expr substr $(uname -s) 1 5)" == "Linux" ]; then
     echo "Platform: Linux"
     # we need libusb headers on circleci
-    yum install -y libusb-devel
+    # yum install -y libusb-devel
     make logger=0 lib/libseabreeze${SHLIB_EXT}
 fi
 cd ../../..
diff --git a/recipe/meta.yaml b/recipe/meta.yaml
index 2d502b9..6e44353 100644
--- a/recipe/meta.yaml
+++ b/recipe/meta.yaml
@@ -16,7 +16,6 @@ requirements:
   build:
     - python
     - setuptools
-    - toolchain
     - cython
   run:
     - python
```
Maybe this should appear in another branch?

ap--, 2017-08-07:
Oh, yes. The recipe in the repository was written for conda-forge which is using circle-ci for compiling the linux packages.
I compiled the armv7 package on raspbian (which is why yum is not there, and the toolchain dependency makes no sense).