ap--/python-seabreeze #47: Unable to use py-seabreeze to access RedTide 650

JuanCab, 2017-05-30:
### spectrometer and system information ###

- model: RedTide 650
- operating system: macOS 10.12.5
- python version:3.5.1
- python-seabreeze version: 0.6.0
- installed-via: _conda

### current problem ###

In a nutshell, the same code I have that works with an Ocean Optics USB2000 when applied to a RedTide 650 (which appears to be listed on the list of compatible devices) doesn't work due to a failure to connect to the spectrometer.  The specific steps are below:

### steps to reproduce ###

Specifically, although Apple's System Profiler clearly shows the RedTide connected (and I can access it via Vernier's Logger Pro software), when I do 

```
import seabreeze.spectrometers as sb
devices = sb.list_devices()
print(devices)
```
I get a blank list.

ap--, 2017-05-31:
With the new conda packages python-seabreeze defaults to the cseabreeze backend.
Maybe USB650 support doesn't work in cseabreeze... If so we need to file this upstream again.

In the meantime try installing python-seabreeze and pyUSB in conda:
```shell
conda install -c poehlmann python-seabreeze
pip install pyusb
```
(Note: the akode channel that we used before to install pyusb only ships it for 2.7. So we'll just install pyusb in conda via pip from pypi)

To check if pyusb was installed correctly run (it should not raise an exception):
```python
import usb.core
usb.core.find()
```

and then:
```python
import seabreeze
seabreeze.use("pyseabreeze")
import seabreeze.spectrometers as sb
devices = sb.list_devices()
print(devices)
```

I hope it'll work. I'll check in the meantime why the 650 isn't working with cseabreeze.

JuanCab, 2017-05-31:
I will try this when I get back in the lab, but one quick question, is there a way to properly check if the cseabreeze interface works (for example, if devices = sb.list_devices() returns no devices) and then switch the interface to pyusb?  I'm going to try an approach, but I suspect I can't simply re-import seabreeze.spectrometers.

ap--, 2017-05-31:
Hmm. I think this might work on OSX and Linux:

```python
# There is definitely a spectrometer connected, and the code
# should first try to connect via cseabreeze, and switch to
# pyseabreeze if it can't find the spectrometer
from seabreeze.backends import _use_cseabreeze
_lib = _use_cseabreeze()
if _lib is not None and _lib.device_list_devices():
    pass  # cseabreeze import works and it finds a spectrometer
else:
    # cseabreeze import failed or it didn't find a spectrometer
    import seabreeze
    seabreeze.use("pyseabreeze")

# continue normally
import seabreeze.spectrometers as sb
...
```

ap--, 2017-05-31:
So as additional info:

I think reimporting `seabreeze.spectrometers` won't work,

but `_use_cseabreeze` and `_use_pyseabreeze` in `seabreeze.backends` return handles that have all the library methods including `device_list_devices()`.
So it should be possible to do the check before importing `seabreeze.spectrometers`

ap--, 2017-05-31:
**Note to self:** Maybe this should be default behavior...

JuanCab, 2017-05-31:
Using your 'auto selection' code I was able to see the USB650, but I am getting errors when attempting to take a spectrum, even turning off dark and non-linearity corrections.  The errors don't occur during every exposure, but happen about 80% of the time.  Looks to be ultimately a USB overflow problem  (based on reviewing error messages).

Juan

```
Traceback (most recent call last):
  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/common.py", line 14, in decorated_func
    return func(*args, **kwargs)
  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/spectrometer.py", line 87, in get_unformatted_spectrum
    timeout=int(self._INTEGRATION_TIME_MAX * 1e-3 + self.usbtimeout_ms))
  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/communication.py", line 65, in usb_read_highspeed
    return self._device.read(self._ENDPOINT_MAP.highspeed_in, size, timeout=timeout)
  File "~/anaconda3/lib/python3.5/site-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "~/anaconda3/lib/python3.5/site-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "~/anaconda3/lib/python3.5/site-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "~/anaconda3/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 84] Overflow
Traceback (most recent call last):
  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/common.py", line 14, in decorated_func
    return func(*args, **kwargs)
  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/spectrometer.py", line 87, in get_unformatted_spectrum
    timeout=int(self._INTEGRATION_TIME_MAX * 1e-3 + self.usbtimeout_ms))
  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/communication.py", line 65, in usb_read_highspeed
    return self._device.read(self._ENDPOINT_MAP.highspeed_in, size, timeout=timeout)
  File "~/anaconda3/lib/python3.5/site-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "~/anaconda3/lib/python3.5/site-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "~/anaconda3/lib/python3.5/site-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "~/anaconda3/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 84] Overflow

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/common.py", line 14, in decorated_func
    return func(*args, **kwargs)
  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/spectrometer.py", line 129, in get_formatted_spectrum
    self.get_unformatted_spectrum(tmp)
  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/common.py", line 23, in decorated_func
    raise SeaBreezeError(msg)
seabreeze.pyseabreeze.interfaces.common.SeaBreezeError: Error while reading raw spectrum.
Traceback (most recent call last):

  File "<ipython-input-17-710a9766b31a>", line 1, in <module>
    runfile('~/Documents/Research/Eclipse 2017/Experiments/Eclipse Software/spec_controller.py', wdir='~/Documents/Research/Eclipse 2017/Experiments/Eclipse Software')

  File "~/anaconda3/lib/python3.5/site-packages/spyder/utils/site/sitecustomize.py", line 880, in runfile
    execfile(filename, namespace)

  File "~/anaconda3/lib/python3.5/site-packages/spyder/utils/site/sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "~/Documents/Research/Eclipse 2017/Experiments/Eclipse Software/spec_controller.py", line 28, in <module>
    lamb, intens = spec.exposure(dev, time, darkcorr=False, nlcorr=False, verbose=True)

  File "~/Documents/Research/Eclipse 2017/Experiments/Eclipse Software/speclib.py", line 134, in exposure
    intens = spec.intensities(correct_dark_counts=darkcorr, correct_nonlinearity=nlcorr)

  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/spectrometers.py", line 150, in intensities
    out[transfered_N:])

  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/wrapper.py", line 123, in spectrometer_get_formatted_spectrum
    return device.interface.get_formatted_spectrum(out)

  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/common.py", line 14, in decorated_func
    return func(*args, **kwargs)

  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/spectrometer.py", line 129, in get_formatted_spectrum
    self.get_unformatted_spectrum(tmp)

  File "~/anaconda3/lib/python3.5/site-packages/seabreeze/pyseabreeze/interfaces/common.py", line 23, in decorated_func
    raise SeaBreezeError(msg)

SeaBreezeError: Error while reading raw spectrum.
```

ap--, 2017-05-31:
Does this also occur when you don't use the "auto-select" code? So just:
```python
import seabreeze
seabreeze.use("pyseabreeze")
import seabreeze.spectrometers as sb
```

And possible ways to proceed are:
1. We fix the seabreeze C library (and therefore cseabreeze)
    1. File an issue on sourceforge, and request USB650 support (Because, while they provide windows drivers and correct udev-rules, it's not implemented in the code)
    2. I can walk you through how to do it yourself if you're comfortable with C/C++.
2. We fix pyseabreeze
    1. I can walk you through how to do it yourself.

Anyways, it would be good to have a datasheet with the full usb command set and compare it in detail with the one for the USB2000 (Because I think it's almost identical to that one)

JuanCab, 2017-05-31:
Not using the auto-select does in fact stabilize things.  I can now perform exposures on the USB650 without issue.  

I am comfortable with C (its been a few years, but it dominated my grad school days), less so with C++.  Feel free to provide me with guidance on how to fix this if by fix you mean allowing auto-select to function.  I think I should file an issue with seabreeze as well, but do you have a specific error to report other than USB 650 support is not functional?

ap--, 2017-06-01:
To just fix the auto-select, I think we don't need to touch any of the C/C++ stuff...

I think my previous approach doesn't work because the cseabreeze backend is not unloaded correctly.

Maybe try:
```python
import usb.core
import seabreeze
if usb.core.find(idVendor=0x2457, idProduct=0x1014):
    # If we find a connected USB650
    seabreeze.use("pyseabreeze")
else:
    seabreeze.use("cseabreeze")

import seabreeze.spectrometers as sb
```

But: Why don't you just use the pyseabreeze backend for all spectrometers?

And regarding the upstream ticket: it's just that USB650 support is missing.

JuanCab, 2017-06-02:
I didn't want to try the new backend because I am in the middle of an experiment and my mentality is to keep everything the same.  That said, I can't imagine there should be a difference, so that may be the easier solution here.

I'll try your solution to the auto-switching shortly.

JuanCab, 2017-06-02:
FYI, I posted a trouble ticket for seabreeze, response was that they intentionally left out RedTide USB650 support. See https://sourceforge.net/p/seabreeze/tickets/32/

JuanCab, 2017-06-02:
On a positive note, your "switching" code above does seem to do the trick.

JuanCab, 2017-06-02:
I take it back, I am still getting USB overflow errors, even when switching to using pyUSB exclusively. :/

ap--, 2017-06-02:
Regarding the sourceforge ticket: Well I guess that explains why I can't find a USB-communication datasheet for the USB650... Weird though that they intentionally leave out support for it...

Regarding the overflow issue: Without the hardware or a datasheet it's hard for me to help with this.
It could be, that some of the assumed parameters for the 650 in [pyseabreeze/interfaces/defines.py](https://github.com/ap--/python-seabreeze/blob/master/seabreeze/pyseabreeze/interfaces/defines.py) or in [pyseabreeze/interfaces/__init__.py](https://github.com/ap--/python-seabreeze/blob/master/seabreeze/pyseabreeze/interfaces/__init__.py#L83-L96) are wrong.

Is your measuring protocol doing some asynchronous stuff? Or something else that might cause the sporadic failures?

JuanCab, 2017-06-02:
I'm not doing anything asynchronous.  Just attaching the spectrometer, taking a reading, and detaching the spectrometer.  And as I said, it works with the Ocean Optics USB2000 even when accessing via pyusb.  I am doing this in iPython, so maybe iPython doesn't close something it should.

ap--, 2017-06-02:
And one last idea:
Maybe the USB650 is not a rebranded USB2000, but actually a rebranded USB4000...
(\<tinfoil hat\> The [usb650](https://oceanoptics.com/product/usb-650-red-tide-spectrometers/) product page has a USB4000 tag... \</tinfoil hat\>)

```python
import seabreeze
seabreeze.use("pyseabreeze")
# Monkey patch the USB650 to use the USB4000 protocol
from seabreeze.pyseabreeze.interfaces import USBInterfaces, USB4000
USBInterfaces[0x1014] = USB4000

import seabreeze.spectrometers as sb
...
```

No idea if this will work. But I guess it's worth one last try...

JuanCab, 2017-06-02:
Pretty sure the USB650 is a rebranded USB2000, it has a sticker on the bottom indicating it is a USB2000.

ap--, 2017-06-02:
Okay... Then I am out of ideas. Sorry.

prissi, 2021-02-17:
The USB650 seems to return the spectrum in a different format.
Each 64 byte block first contains the low bytes and the next block 64 block the high bytes.
So two data points are actually constructed like this:

So for each 64 block one has to do: (j contains the offset in the raw data, i the spectrum point)
data[i] = raw[j+1]+256*raw[j+1+64]
data[i+1] = raw[j]+256*raw[j+64]
i += 2;
j += 2;

ap--, 2021-02-17:
Hi @prissi 

The current device implementation in the `pyseabreeze` backend is based on the `USB2000` (not `USB2000+`) implementation.

https://github.com/ap--/python-seabreeze/blob/ff806a834019e2d5a548d2af629d01c90646fea8/src/seabreeze/pyseabreeze/devices.py#L652-L678

which uses the millisecond timescale the smaller max value and the following raw spectrum formatting:

https://github.com/ap--/python-seabreeze/blob/ff806a834019e2d5a548d2af629d01c90646fea8/src/seabreeze/pyseabreeze/features/spectrometer.py#L164-L175

does this agree with your comment?

Cheers,
Andreas 😃 

P.S.: Although I still am not entirely certain that the USB650 works with python-seabreeze, I believe that some USB650 require a firmware update to be supported. See: https://github.com/ap--/python-seabreeze/issues/48
It would be super interesting to know the firmware version of USB650 spectrometers and a report if they work with the pyseabreeze backend or not.

prissi, 2021-02-18:
My knowledge of python and numpy is not good enough to answer that. But it seems similar. Seabreeze does not install on this machine, neither from Mingw python (where already numpy does not work) or doesnot work  from anaconda. But I am not python specialist.

The USB650 is a recent model. OceanView does not report a firmware version, so I am not sure how to check for the firmware version.

Since the target computer is an XP without network access, I am using teh USB650 with the libusb from C directly and it works like an existing USB2000+, but with the different byte order as above. I arrived (at the old repo), because this is one of the very few places that had any info on the USB650, and the documentation here was enough to get it working.

closed 2023-06-11 by ap--

rusbus123, 2025-05-23:
@prissi could you provide the script that worked for the USB650? Would be very much appreciated!

prissi, 2025-05-24:
I have never used the python lib, I use my own C code.
The USB650 has these differences
1) Integration time must be sent in ms not microseconds as most of the others
2) The data are in blocks of 16 like the original USB2000 (I think)
3) No support for temperature readout
Maybe using the USB2000 for it would work.

In C code ...
USB HD2000 and 650 have the same identification
(desc.idVendor == 0x2457 && desc.idProduct == 0x1014)
The ports are the same too:
#define EP_IN1 (0x87) // USB650
#define EP_IN2 (0x82)
static int EP_OUT = 2;   /// USB650
static int EP_OUT2 = 7;   /// USB650, not used for simple queries

Then try to get CCD temperature to find out if USB650
libusb_bulk_transfer(devFlame, EP_OUT, (unsigned char*)"\x6C", 1, NULL, 100);
if (libusb_bulk_transfer(devFlame, EP_IN1, buffer, 3, NULL, 100) < 0) {
  // USB650
  ...
}
If USB650 then timeconst are from 3 ms (3000 us) to 15000 ms (15000000 us)! (So 1000 times less than all other)

Reading data is a little convoluted
#define MAX_DATA (2048)
#define SKIP_START (24)
const int num_pts = MAX_DATA - SKIP_START; // first 24 are zero
memset(data, 0, sizeof(uint32_t) * MAX_DATA);
memset(spec.data, 0, sizeof(double) * MAX_DATA);
DWORD timeout = (last_time_constant/900+100);

error = libusb_bulk_transfer(devFlame, EP_OUT, (unsigned char*)"\x09", 1, &num_bytes, timeout);
if (!error) error = libusb_bulk_transfer(devFlame, EP_IN2, buffer, 4097, &num_bytes, timeout);
if (error || num_bytes < 4097) {
	while (!libusb_bulk_transfer(devFlame, EP_IN2, buffer, 4097, &num_bytes, 100));
	while (!libusb_bulk_transfer(devFlame, EP_IN1, buffer, 4097, &num_bytes, 100));
	return 0;
}
uint16_t max_value = 0;
for (int i = SKIP_START; i < SKIP_START + num_pts; ) {
	int offset = (i / 64) * 128;
	int start_k = (i % 64);
	for (int k = start_k; k < 64; k++) {
		uint16_t d = buffer[offset + k] + 256 * buffer[offset + k + 64];
		data[i++] = d;
		if (d > max_value) {
			max_value = d;
		}
	}
}
spec.temperature = 0;

rusbus123, 2025-05-26:
@prissi Thanks!

PR #198 (2023-06-11): Dynamically limit max ep size
This should fix overflowerrors as in:

Closes #190
Closes #193
Closes #174
Closes #47

Replaces #183