ap--/python-seabreeze #48: USB2000 stuck in infinite loop when asking for spec.intensities

aoabdelatif, 2017-06-06:
Hey ap--

im new to python (experienced C and arduino user here).

anyways, im currently trying to get the intensity values from a USB2000 ocean optics device.

currently, it seems like everything is working. USB 2000 is detected, wavelengths seem correct and printed. 

The problem is when i call spec.intensities, my command line goes into an infinite loop. 

heres my code...

import seabreeze.spectrometers as sb
devices = sb.list_devices()
print(devices)
spec = sb.Spectrometer(devices[0])
spec.integration_time_micros(10000)
Wavelengths = spec.wavelengths()
print(Wavelengths)
value = spec.intensities()
print(value)

the command line just remains blank after it runs

im running Spyder from anaconda

image of command line

[image]

aoabdelatif, 2017-06-06:
Update, after I left the program running for a long time, it finally gave me this error

[image]

aoabdelatif, 2017-06-06:
Mac OSX Sierra
Python 3
USB2000 Spectrometer
Spyder Anaconda

ap--, 2017-06-06:
Hey aoabdelatif,

can you try the same outside of spyder in a python interpreter and report back?

All the best,
-Andreas

aoabdelatif, 2017-06-06:
Hey Andreas,

Thank you for the speedy response.

So would you recommend using IDLE?

OSX Sierra
Python 3

-Adam

ap--, 2017-06-06:
Just open your anaconda python interpreter from the command line (Terminal).
(open your terminal app and type `python`. Note: it should say something like python 3.6.1 anaconda..., if not, you need to [export your Anaconda path](https://docs.continuum.io/anaconda/install-macos))

It's possible that spyder or idle do some extra things, that might interfere.
If the problem is still there in a plain python interpreter, we need to investigate further.

aoabdelatif, 2017-06-07:
Andreas,

Here is a screen shot of what my terminal looked like after i ran the same functions as i did in spyder. 

Once used the spec.intensities() function, the same thing happened as before, the terminal line got hung up once again. (Ive been waiting for awhile and its still hung up)

[image]

MacOSX Sierra 10.12.1
Python 3.6.1
Anaconda 4.4.0
USB2000

I really appreciate your help with all this,

-Adam

ps. 
I hope this doesn't turn you off from helping me but i also downloaded the OceanView Free Trial and for some reason OceanView gets hung up when i try to connect to the USB2000, Ocean View does recognize the USB2000.

ap--, 2017-06-07:
Hmm, this is very strange.
It should return immediately...

can you run this in your terminal:
```
pip install pyusb
```
then run python and run:
```python
import seabreeze
seabreeze.use('pyseabreeze')
import seabreeze.spectrometers as sb
devices = sb.list_device()
spec = sb.Spectrometer(devices[0])
spec.intensities()
```

aoabdelatif, 2017-06-10:
Hello Andreas,

I think you can go ahead and close this issue.

With a lot of trouble shooting, blood, sweat, and tears, i finally found the fix!

It turns out, my USB2000 Ocean Optic Spectrometer never had the firmware updated.

The firmware is required in order to run it on Mac, Windows 8, and Windows 10.

The only way to get the firmware is to email Ocean Optics Technical Support and they will send you the update file with instructions on how to install the firmware on to your device.

I successfully installed the firmware and now my spectrometer runs perfect!

Thank you for all your help!

-Adam

ap--, 2017-06-11:
HI Adam,
Thanks for the report. I'll leave the issue open, until I add this info to the README at some point.
Glad that it works now :)
-Andreas

aliakbarro, 2024-12-29:
> Hello Andreas,
>
> I think you can go ahead and close this issue.
>
> With a lot of trouble shooting, blood, sweat, and tears, i finally found the fix!
>
> It turns out, my USB2000 Ocean Optic Spectrometer [...]

Hey. can you share the files with me? I have the same problem