ap--/python-seabreeze #51: Unable to reload udev rules on Raspberry Pi 3

benjaminesse, 2017-07-10:
### spectrometer and system information ###

- model: USB2000+
- operating system: Raspbian 'jessie'
- python version: 3.6.1
- python-seabreeze version: current master
- installed-via: conda (Berryconda)

### current problem ###

Installation of python-seabreeze is fine, but when I try to reset the rules in /etc/udev/rules.d/ by running:
```shell
sudo udevadm --reload-rules
```
it returns 
```shell
udevadm: unrecognized option '--reload-rules'
```
After googling I haven't found much to help, I am new to the Raspberry Pi. I have tried 

```shell
sudo udevadm control --reload-rules
```

which executes with no error but doesn't allow the spectrometer to be detected.

I have tried several reboots and unplugs/replugs of the spectrometer.

### steps to reproduce ###

_Step by step instructions to reproduce the error. The more detailed the better, but please use some common sense:_

1. Download 10-oceanoptics.rules and copy to /etc/udev/rules.d/
2. Run 
```shell
sudo udevadm --reload-rules
```
3. Returns message: 
```shell
udevadm: unrecognized option '--reload-rules'
```

In python, running 
```python
import seabreeze.spectrometers as sb
sb.list_devices()
```
returns
```python
[]
```
with no spectrometers attached (as expected) and
```python
usb_claim_interface() returned -1 - did you copy os-support/linux/10-oceanoptics.rules to /etc/udev/rules.d?
Could not open device with ID 0. error=-1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "seabreeze/cseabreeze/wrapper.pyx", line 84, in seabreeze.cseabreeze.wrapper.device_list_devices (./seabreeze/cseabreeze/wrapper.c:3750)
  File "seabreeze/cseabreeze/wrapper.pyx", line 133, in seabreeze.cseabreeze.wrapper.device_get_serial_number (./seabreeze/cseabreeze/wrapper.c:4711)
seabreeze.cseabreeze.wrapper.SeaBreezeError: This should not have happened. Apparently this device has 0 serial number features. The code expects it to have 1 and only 1. Please file a bug report including a description of your device.
```
with the USB2000+ connected.

Any light you can shed would be gratefully received.

Many thanks!

ap--, 2017-07-21:
Hi @benjaminesse,

sorry for the long delay.
What happens when you run your python code as root with a spectrometer attached?

-Andreas

benjaminesse, 2017-07-24:
Hi,

That works fine, it finds the spectrometer with no problems. Is there a way around running it from root every time? I'm not too familiar with Linux, I have used it sporadically but most of my development has been in windows, so apologies if I'm missing something!

Cheers,

Ben

ap--, 2017-07-24:
See if your user is in the "plugdev" group:
[see which groups you linux user belongs to](https://www.howtogeek.com/howto/ubuntu/see-which-groups-your-linux-user-belongs-to/)

And create the "plugdev" group and add your user to it if necessary:
[add a user to a group](https://www.howtogeek.com/50787/add-a-user-to-a-group-or-second-group-on-linux/) 

-Andreas

soumendras, 2017-07-25:
Hey wait...i have the exact same problem ....i did not understand how benjamin solved his problem ...please help me solve it ...... and AP ....when i write groups in my pi user name it returns plugdev

ap--, 2017-07-25:
your udev rules are installed correctly if there is a new symlink created in `/dev` after you plug in your spectrometer. This symlink (if you have a USB2000 spectrometer, for example: `/dev/usb2000-12345`, where 12345 is some number or string) should be owned by the plugdev group.

benjaminesse, 2017-07-26:
Hi,

I reloaded the udev rules fine with the update you referenced above, but I still can't connect without running the code from root.

I checked /dev and couldn't find a symlink (I'm running a Flame so I'm assuming it will be of the form "FLM-12345")

Thanks again

ap--, 2017-07-26:
I think the flame should show up as a usb2000plus-

what's the output of lsusb when spectrometer is connected?

benjaminesse, 2017-07-26:
There is no USB symlink either. The output of lssub is:
```bash
Bus 001 Device 006: ID 2457:101e  
Bus 001 Device 005: ID 03f0:034a Hewlett-Packard 
Bus 001 Device 004: ID 0461:4d22 Primax Electronics, Ltd 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

benjaminesse, 2017-07-26:
It returns
```bash
ls: cannot access /dev/usb*: No such file or directory
```

ap--, 2017-07-26:
and 
```
ls /etc/udev/rules.d/
```

benjaminesse, 2017-07-26:
```bash
10-oceanoptics.rules 99-com.rules
```

ap--, 2017-07-26:
Hmm,
this is most likely due to your udev version.
Try using an older version of the rules from [here](https://raw.githubusercontent.com/ap--/python-seabreeze/5d882443720b8dfc5be514ad2d10374a8799228c/misc/10-oceanoptics.rules)

download the file linked to above and overwrite the file `/etc/udev/rules.d/10-oceanoptics.rules` on your rpi.

make sure that when you type `cat /etc/udev/rules.d/10-oceanoptics.rules` it says ATTRS everywhere and not ATTR (as it does in the newer version of that file)

after that reboot your rpi.

If this doesn't help, I can't really provide further help.
It's best if you google how to debug udev rules and try to figure out why the provided rules don't match. It's not complicated to do that. You'll find a tutorial easily. Good luck!

benjaminesse, 2017-07-26:
Nope, still got the same problem. I'll look into the udev rules then, thanks for your help!

ap--, 2017-07-26:
Np.

one more quick thing to try:
use the newer file again (the one with ATTR)
and delete line 4, the one that starts with SUBSYSTEM...
save it.
reboot.

If that didn't help, and you figure out what's wrong, it would be great if you could report back!
Good luck!

soumendras, 2017-07-27:
when i try to run sudo python ......i cannot import this "import seabreeze.spectrometers as sb" .....says no such module

ap--, 2017-07-27:
@soumendras https://stackoverflow.com/questions/25346171/cant-import-module-when-using-root-user

ap--, 2017-07-27:
@benjaminesse 
run `lsusb`
it returns something like:
```
Bus 001 Device 006: ID 2457:101e  
Bus 001 Device 005: ID 03f0:034a Hewlett-Packard 
Bus 001 Device 004: ID 0461:4d22 Primax Electronics, Ltd 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`2457:101e` is your spectrometer. Remember Bus 001 and Device 006. and use these numbers in the following command:

```
udevadm info --attribute-walk --name=/dev/bus/usb/001/006
```

post the output here.

soumendras, 2017-07-27:
in my case the above command returns this 

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5':
    KERNEL=="1-1.5"
    SUBSYSTEM=="usb"
    DRIVER=="usb"
    ATTR{bDeviceClass}=="00"
    ATTR{manufacturer}=="USB4000 3.00.03"
    ATTR{bmAttributes}=="80"
    ATTR{bConfigurationValue}=="1"
    ATTR{version}==" 2.00"
    ATTR{devnum}=="6"
    ATTR{bMaxPower}=="400mA"
    ATTR{idProduct}=="1022"
    ATTR{avoid_reset_quirk}=="0"
    ATTR{urbnum}=="9"
    ATTR{bDeviceSubClass}=="00"
    ATTR{maxchild}=="0"
    ATTR{bcdDevice}=="0002"
    ATTR{bMaxPacketSize0}=="64"
    ATTR{idVendor}=="2457"
    ATTR{product}=="Ocean Optics USB4000"
    ATTR{speed}=="480"
    ATTR{removable}=="removable"
    ATTR{ltm_capable}=="no"
    ATTR{bNumConfigurations}=="1"
    ATTR{busnum}=="1"
    ATTR{authorized}=="1"
    ATTR{quirks}=="0x0"
    ATTR{configuration}==""
    ATTR{devpath}=="1.5"
    ATTR{bDeviceProtocol}=="00"
    ATTR{bNumInterfaces}==" 1"

  looking at parent device '/devices/platform/soc/3f980000.usb/usb1/1-1':
    KERNELS=="1-1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{bDeviceClass}=="09"
    ATTRS{bmAttributes}=="e0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{version}==" 2.00"
    ATTRS{devnum}=="2"
    ATTRS{bMaxPower}=="2mA"
    ATTRS{idProduct}=="9514"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{urbnum}=="69"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{maxchild}=="5"
    ATTRS{bcdDevice}=="0200"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{idVendor}=="0424"
    ATTRS{speed}=="480"
    ATTRS{removable}=="unknown"
    ATTRS{ltm_capable}=="no"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{busnum}=="1"
    ATTRS{authorized}=="1"
    ATTRS{quirks}=="0x0"
    ATTRS{configuration}==""
    ATTRS{devpath}=="1"
    ATTRS{bDeviceProtocol}=="02"
    ATTRS{bNumInterfaces}==" 1"

  looking at parent device '/devices/platform/soc/3f980000.usb/usb1':
    KERNELS=="usb1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{bDeviceClass}=="09"
    ATTRS{manufacturer}=="Linux 4.9.35-v7+ dwc_otg_hcd"
    ATTRS{bmAttributes}=="e0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{version}==" 2.00"
    ATTRS{devnum}=="1"
    ATTRS{bMaxPower}=="0mA"
    ATTRS{idProduct}=="0002"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{urbnum}=="25"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{maxchild}=="1"
    ATTRS{bcdDevice}=="0409"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{idVendor}=="1d6b"
    ATTRS{product}=="DWC OTG Controller"
    ATTRS{speed}=="480"
    ATTRS{authorized_default}=="1"
    ATTRS{interface_authorized_default}=="1"
    ATTRS{removable}=="unknown"
    ATTRS{ltm_capable}=="no"
    ATTRS{serial}=="3f980000.usb"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{busnum}=="1"
    ATTRS{authorized}=="1"
    ATTRS{quirks}=="0x0"
    ATTRS{configuration}==""
    ATTRS{devpath}=="0"
    ATTRS{bDeviceProtocol}=="01"
    ATTRS{bNumInterfaces}==" 1"

  looking at parent device '/devices/platform/soc/3f980000.usb':
    KERNELS=="3f980000.usb"
    SUBSYSTEMS=="platform"
    DRIVERS=="dwc_otg"
    ATTRS{wr_reg_test}=="Time to write GNPTXFSIZ reg 10000000 times: 730 msecs (                                                                             73 jiffies)"
    ATTRS{grxfsiz}=="GRXFSIZ = 0x00000306"
    ATTRS{srpcapable}=="SRPCapable = 0x1"
    ATTRS{buspower}=="Bus Power = 0x1"
    ATTRS{bussuspend}=="Bus Suspend = 0x0"
    ATTRS{hptxfsiz}=="HPTXFSIZ = 0x02000406"
    ATTRS{hnp}=="HstNegScs = 0x0"
    ATTRS{mode}=="Mode = 0x1"
    ATTRS{mode_ch_tim_en}=="Mode Change Ready Timer Enable = 0x0"
    ATTRS{hsic_connect}=="HSIC Connect = 0x1"
    ATTRS{gsnpsid}=="GSNPSID = 0x4f54280a"
    ATTRS{driver_override}=="(null)"
    ATTRS{hcd_frrem}=="HCD Dump Frame Remaining"
    ATTRS{gotgctl}=="GOTGCTL = 0x001c0001"
    ATTRS{gpvndctl}=="GPVNDCTL = 0x00000000"
    ATTRS{hnpcapable}=="HNPCapable = 0x1"
    ATTRS{spramdump}=="SPRAM Dump"
    ATTRS{regoffset}=="0xffffffff"
    ATTRS{gnptxfsiz}=="GNPTXFSIZ = 0x01000306"
    ATTRS{guid}=="GUID = 0x2708a000"
    ATTRS{regdump}=="Register Dump"
    ATTRS{hprt0}=="HPRT0 = 0x00001005"
    ATTRS{hcddump}=="HCD Dump"
    ATTRS{rem_wakeup_pwrdn}==""
    ATTRS{regvalue}=="invalid offset"
    ATTRS{gusbcfg}=="GUSBCFG = 0x20001700"
    ATTRS{fr_interval}=="Frame Interval = 0x1d4b"
    ATTRS{busconnected}=="Bus Connected = 0x1"
    ATTRS{remote_wakeup}=="Remote Wakeup Sig = 0 Enabled = 0 LPM Remote Wakeup =                                                                              0"
    ATTRS{devspeed}=="Device Speed = 0x0"
    ATTRS{rd_reg_test}=="Time to read GNPTXFSIZ reg 10000000 times: 1410 msecs (                                                                             141 jiffies)"
    ATTRS{enumspeed}=="Device Enumeration Speed = 0x1"
    ATTRS{inv_sel_hsic}=="Invert Select HSIC = 0x0"
    ATTRS{ggpio}=="GGPIO = 0x00000000"
    ATTRS{srp}=="SesReqScs = 0x1"

  looking at parent device '/devices/platform/soc':
    KERNELS=="soc"
    SUBSYSTEMS=="platform"
    DRIVERS==""
    ATTRS{driver_override}=="(null)"

  looking at parent device '/devices/platform':
    KERNELS=="platform"
    SUBSYSTEMS==""
    DRIVERS==""

soumendras, 2017-07-27:
@benjaminesse any update on how did you run python as sudo and still call those dependencies?
@ap-- that link tutorial did not work in my case ...by the way ..did PYTHONPATH is what i have to type or it will be the real path where python is located?

benjaminesse, 2017-07-27:
Hi @soumendras, from that tutorial (and anyone correct me if I am wrong) when you run python from root the normal version associated that is used when you use the command
```bash
python hello.py
```
is not necessarily the same, so any libraries you have installed (such as the python-seabreeze) aren't installed in that version. I got around this by putting
```python
#!PATHTOPYTHON
```
as the first line in the program replacing the PATHTOPYTHON with the actual path eg
```python
#!/home/pi/berryconda3/bin/python3.6
```
for me. You then run it from root by typing
```bash
sudo ./hello.py
```
in the command line.

@ap-- I tried your suggestion of removing the lines and it worked, I now have no issues getting the program to detect the spectrometer, thank you! The output of the command you posted 2 comments back is:
```bash
Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5':
    KERNEL=="1-1.5"
    SUBSYSTEM=="usb"
    DRIVER=="usb"
    ATTR{bDeviceSubClass}=="00"
    ATTR{bDeviceProtocol}=="00"
    ATTR{devpath}=="1.5"
    ATTR{idVendor}=="2457"
    ATTR{speed}=="480"
    ATTR{bNumInterfaces}==" 1"
    ATTR{bConfigurationValue}=="1"
    ATTR{bMaxPacketSize0}=="64"
    ATTR{busnum}=="1"
    ATTR{devnum}=="7"
    ATTR{configuration}==""
    ATTR{bMaxPower}=="400mA"
    ATTR{authorized}=="1"
    ATTR{bmAttributes}=="80"
    ATTR{bNumConfigurations}=="1"
    ATTR{maxchild}=="0"
    ATTR{bcdDevice}=="0002"
    ATTR{avoid_reset_quirk}=="0"
    ATTR{quirks}=="0x0"
    ATTR{version}==" 2.00"
    ATTR{urbnum}=="64"
    ATTR{ltm_capable}=="no"
    ATTR{manufacturer}=="Flame-S 4.01.0 "
    ATTR{removable}=="removable"
    ATTR{idProduct}=="101e"
    ATTR{bDeviceClass}=="00"
    ATTR{product}=="Ocean Optics Flame-S"

  looking at parent device '/devices/platform/soc/3f980000.usb/usb1/1-1':
    KERNELS=="1-1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bDeviceProtocol}=="02"
    ATTRS{devpath}=="1"
    ATTRS{idVendor}=="0424"
    ATTRS{speed}=="480"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{busnum}=="1"
    ATTRS{devnum}=="2"
    ATTRS{configuration}==""
    ATTRS{bMaxPower}=="2mA"
    ATTRS{authorized}=="1"
    ATTRS{bmAttributes}=="e0"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{maxchild}=="5"
    ATTRS{bcdDevice}=="0200"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{quirks}=="0x0"
    ATTRS{version}==" 2.00"
    ATTRS{urbnum}=="112"
    ATTRS{ltm_capable}=="no"
    ATTRS{removable}=="unknown"
    ATTRS{idProduct}=="9514"
    ATTRS{bDeviceClass}=="09"

  looking at parent device '/devices/platform/soc/3f980000.usb/usb1':
    KERNELS=="usb1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bDeviceProtocol}=="01"
    ATTRS{devpath}=="0"
    ATTRS{idVendor}=="1d6b"
    ATTRS{speed}=="480"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{authorized_default}=="1"
    ATTRS{busnum}=="1"
    ATTRS{devnum}=="1"
    ATTRS{configuration}==""
    ATTRS{bMaxPower}=="0mA"
    ATTRS{authorized}=="1"
    ATTRS{bmAttributes}=="e0"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{maxchild}=="1"
    ATTRS{interface_authorized_default}=="1"
    ATTRS{bcdDevice}=="0404"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{quirks}=="0x0"
    ATTRS{serial}=="3f980000.usb"
    ATTRS{version}==" 2.00"
    ATTRS{urbnum}=="25"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Linux 4.4.34-v7+ dwc_otg_hcd"
    ATTRS{removable}=="unknown"
    ATTRS{idProduct}=="0002"
    ATTRS{bDeviceClass}=="09"
    ATTRS{product}=="DWC OTG Controller"

  looking at parent device '/devices/platform/soc/3f980000.usb':
    KERNELS=="3f980000.usb"
    SUBSYSTEMS=="platform"
    DRIVERS=="dwc_otg"
    ATTRS{hnp}=="HstNegScs = 0x0"
    ATTRS{srp}=="SesReqScs = 0x1"
    ATTRS{regvalue}=="invalid offset"
    ATTRS{hsic_connect}=="HSIC Connect = 0x1"
    ATTRS{guid}=="GUID = 0x2708a000"
    ATTRS{mode}=="Mode = 0x1"
    ATTRS{srpcapable}=="SRPCapable = 0x1"
    ATTRS{regdump}=="Register Dump"
    ATTRS{gpvndctl}=="GPVNDCTL = 0x00000000"
    ATTRS{ggpio}=="GGPIO = 0x00000000"
    ATTRS{hprt0}=="HPRT0 = 0x00001005"
    ATTRS{wr_reg_test}=="Time to write GNPTXFSIZ reg 10000000 times: 370 msecs (37 jiffies)"
    ATTRS{driver_override}=="(null)"
    ATTRS{hcd_frrem}=="HCD Dump Frame Remaining"
    ATTRS{mode_ch_tim_en}=="Mode Change Ready Timer Enable = 0x0"
    ATTRS{gnptxfsiz}=="GNPTXFSIZ = 0x01000306"
    ATTRS{remote_wakeup}=="Remote Wakeup Sig = 0 Enabled = 0 LPM Remote Wakeup = 0"
    ATTRS{busconnected}=="Bus Connected = 0x1"
    ATTRS{hcddump}=="HCD Dump"
    ATTRS{gotgctl}=="GOTGCTL = 0x001c0001"
    ATTRS{spramdump}=="SPRAM Dump"
    ATTRS{grxfsiz}=="GRXFSIZ = 0x00000306"
    ATTRS{gsnpsid}=="GSNPSID = 0x4f54280a"
    ATTRS{gusbcfg}=="GUSBCFG = 0x20001700"
    ATTRS{hptxfsiz}=="HPTXFSIZ = 0x02000406"
    ATTRS{devspeed}=="Device Speed = 0x0"
    ATTRS{fr_interval}=="Frame Interval = 0x1d4c"
    ATTRS{rem_wakeup_pwrdn}==""
    ATTRS{bussuspend}=="Bus Suspend = 0x0"
    ATTRS{buspower}=="Bus Power = 0x1"
    ATTRS{hnpcapable}=="HNPCapable = 0x1"
    ATTRS{rd_reg_test}=="Time to read GNPTXFSIZ reg 10000000 times: 940 msecs (94 jiffies)"
    ATTRS{enumspeed}=="Device Enumeration Speed = 0x1"
    ATTRS{inv_sel_hsic}=="Invert Select HSIC = 0x0"
    ATTRS{regoffset}=="0xffffffff"

  looking at parent device '/devices/platform/soc':
    KERNELS=="soc"
    SUBSYSTEMS=="platform"
    DRIVERS==""
    ATTRS{driver_override}=="(null)"

  looking at parent device '/devices/platform':
    KERNELS=="platform"
    SUBSYSTEMS==""
    DRIVERS==""
```

ap--, 2017-07-27:
@benjaminesse 
awesome! I'll fix this at some point in the future, until then we'll leave this open for people who run into the udev issue.

**Note to self:** udev SUBSYSTEM needs to match against "usb" not "usb_device" on raspbian jessie ?!?

### WORKAROUND

**if you're raspbian and you have this issue open the [10-oceanoptics.rules](https://github.com/ap--/python-seabreeze/blob/master/misc/10-oceanoptics.rules#L4) file and replace "usb_device" with "usb" in line 4.**

closed 2017-08-12 by ap--

gabriel-cruz, 2019-06-12:
I had the same problem as this, but my rpi3 does not even recognize the spectrometer when connected, in the lsusb command it is also not listed. I followed all these steps and the tutorial and nothing has changed, what do I do? The model of the spectrometer is the USB2000 +

soumendras, 2019-06-12:
*if you're raspbian and you have this issue open the 10-oceanoptics.rules
<https://github.com/ap--/python-seabreeze/blob/master/misc/10-oceanoptics.rules#L4>
file
and replace "usb_device" with "usb" in line 4.*

*This is what you might be missing. Did u do this? *
*********Quitters never win, Winners never quit**********

SOUMENDRA SINGH
SCIENTIST,
CENTER FOR ASTROPARTICLE PHYSICS & SPACE SCIENCE
BOSE INSTITUTE
BLOCK EN, SECTOR-V
SALT LAKE CITY,
KOLKATA-91

+919432926954
+919433339275

gabriel-cruz, 2019-06-12:
Yes, I did it, it's like this: SUBSYSTEM!="usb", ACTION!="add", GOTO="oceanoptics_rules_end". Is that correct?

soumendras, 2019-06-12:
It is like, backspace from e of device to underscore. Just look for
usb_device in the whole document and replace it by usb.

gabriel-cruz, 2019-06-17:
I searched the document, but there was no other line containing "usb_device"

soumendras, 2019-06-17:
Are you sure u r calling the right python. Remember python and sudo python
will be calling different versions. Make sure u call the one where u
installed miniconda.

gabriel-cruz, 2019-06-19:
I'm not sure what version of python I've installed berryconda, but I'll install it all again to make sure. Apparently the pyseabreeze is installed correctly because when I squeeze the code to look for the device it returns only empty ([]) and throws no exceptions

commit dfc7ba6e (2017-08-12): udev rules: match usb_device or usb

This should fix #51.