ap--/python-seabreeze #53: install_libseabreeze.sh build failure with Log.cpp

petewall, 2017-09-13:
### spectrometer and system information ###

- model: STS-VIS
- operating system: 2017-08-16-raspbian-stretch-lite on a Raspberry Pi Zero
- python version: 2.7.13
- python-seabreeze version: v0.6.0
- installed-via: `install_libseabreeze.sh`

### current problem ###
The Linux build fails on the `install_libseabreeze.sh` step.  The cause is the Seabreeze file Log.cpp that mixes tabs and spaces.  Because of this, the compiler throws this warning, which is treated like an error:

    building Log.o
    Log.cpp: In member function 'void Log::formatAndSend(int, const char*, const char*, const char*, va_list)':
    Log.cpp:163:5: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
         if (logFile == NULL)
         ^~   
    Log.cpp:166:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'if' 
      unsigned indent = (unsigned int) (callstack->size() - 1) * 4; 
      ^~~~~~~~
    Log.cpp:180:5: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
         if (fmt[strlen(fmt)] != '\n')
         ^~   
    Log.cpp:182:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'if' 
      fflush(logFile);
      ^~~~~~
    cc1plus: all warnings being treated as errors
    ../../common.mk:158: recipe for target 'Log.o' failed
    make[2]: *** [Log.o] Error 1
    make[2]: Leaving directory '/home/pi/python-seabreeze-python-seabreeze-v0.6.0/misc/libseabreeze/SeaBreeze/src/common'
    ../common.mk:169: recipe for target 'subdirs' failed
    make[1]: *** [subdirs] Error 2
    make[1]: Leaving directory '/home/pi/python-seabreeze-python-seabreeze-v0.6.0/misc/libseabreeze/SeaBreeze/src'
    Makefile:29: recipe for target 'lib/libseabreeze.so' failed
    make: *** [lib/libseabreeze.so] Error 2

### steps to reproduce ###

With a fresh install of Raspbian:

    sudo apt-get update
    sudo apt-get upgrade
    sudo apt-get install git-all build-essential libusb-dev python-pip

    wget https://github.com/ap--/python-seabreeze/archive/python-seabreeze-v0.6.0.tar.gz
    tar xzvf python-seabreeze-v0.6.0.tar.gz
    cd python-seabreeze-python-seabreeze-v0.6.0/misc
    ./install_libseabreeze.sh

### minimal code example and error (very helpful if available) ###

I did manage to find two possible workarounds:
1. I used `export CFLAGS="-Wno-error=misleading-indentation"` to not trigger that compiler error.  This worked on one of my Raspbian installs, but not another.
2. After I got the failure, I edited Log.cpp, replaced all tab characters (`\t`) with four spaces, then commented out the code that would re-fetch the source:

```diff
> #curl -L -o libseabreeze.zip ${SB_URL}
> #unzip libseabreeze.zip
> #mv libseabreeze-libseabreeze-${SB_VER} libseabreeze
---
< curl -L -o libseabreeze.zip ${SB_URL}
< unzip libseabreeze.zip
< mv libseabreeze-libseabreeze-${SB_VER} libseabreeze
```

Then a rebuild will work.

petewall, 2017-09-13:
I realize that the code that needs to be fixed (Log.cpp) is Ocean Optics own code, not this source, but until they fix it, I could see adding a patch file or adding the CFLAGS parameter until they do.

ap--, 2017-09-20:
we'll wait for upstream to fix this. Until then this stays open

ap--, 2019-09-11:
With version 1.0.0 this is not an issue anymore, because the stripped down libSeabreeze code is vendored with python-seabreeze. :tada:

Installing from source should just work via:
```bash
git clone git@github.com:ap--/python-seabreeze.git
cd python-seabreeze
pip install .
```

see:
https://python-seabreeze.readthedocs.io/en/latest/install.html#building-seabreeze-manually

closed 2019-09-11 by ap--