ap--/python-seabreeze #54: Support for Pixel Binning

petewall, 2017-09-18:
I would love to play with the pixel binning factors to see how that would affect integration time and granularity.

How difficult would it be to add support for these functions in the Seabreeze API:

    Response GetPixelBinningFactor(const std::string &arguments);
    Response SetPixelBinningFactor(const std::string &arguments);
    Response GetDefaultPixelBinningFactor(const std::string &arguments);
    Response SetDefaultPixelBinningFactor(const std::string &arguments);
    Response GetMaxPixelBinningFactor(const std::string &arguments);

I thought about trying to implement it myself, but I worry about running into trouble with the ```_pixels``` field in ```spectrometers.py```

ap--, 2017-09-20:
Hi Pete,

it's not really complicated. You would need to look at the cseabreeze code and wrap the following functions:
```
sbapi_get_number_of_pixel_binning_features(long deviceID, int *error_code);
sbapi_get_pixel_binning_features(long deviceID, int *error_code, long *features,
sbapi_binning_set_pixel_binning_factor(long deviceID, long featureID,
sbapi_binning_get_pixel_binning_factor(long deviceID, long featureID, int *error_code);
sbapi_binning_set_default_pixel_binning_factor(long deviceID, long featureID,
sbapi_binning_reset_default_pixel_binning_factor(long deviceID, long featureID,
sbapi_binning_get_default_pixel_binning_factor(long deviceID, long featureID, int *error_code);
sbapi_binning_get_max_pixel_binning_factor(long deviceID, long featureID, int *error_code);
```
Then you would need to add a (maybe a little more pythonic) interface for the pixel binning to the spectrometer class.

Please note that the seabreeze library only supports pixel binning for the STS (if I'm not mistaken).

All the best,
-Andreas

ap--, 2019-09-11:
With version 1.0.0 all supported SeaBreeze features have been wrapped :tada:

```python
spec = Spectrometer.from_first_available()
spec.f.pixel_binning.set_binning_factor(2)  # etc..
```

see:
https://python-seabreeze.readthedocs.io/en/latest/backend_api.html#seabreeze.cseabreeze.SeaBreezePixelBinningFeature

closed 2019-09-11 by ap--