ap--/python-seabreeze #55: Support For flameNIR

island-monkey, 2017-12-20:
Will there be any support for the flameNIR?

ap--, 2017-12-29:
Hey @island-monkey,

What's the USB product ID of the flameNIR? If I recall correctly (I do have to check again though) the flame spectrometer uses the same protocol as the USB2000+ so it might be supported already.

Best,
Andreas

island-monkey, 2018-01-11:
Hi @ap-- ,

the product ID is : 2457

ap--, 2018-01-12:
that's the vendorID. the other four digit hex number is the productID :)

island-monkey, 2018-01-23:
Ups, sorry :) Its 104b

ap--, 2018-01-31:
Not yet supported by the pyseabreeze backend.
But it should work with the cseabreeze backend.

maykef, 2018-03-21:
It does work:
print device gives:
SeaBreezeDevice USB2000PLUS:FLMS04727 (Flame NIR)

island-monkey, 2018-03-26:
Hi @maykef, are you using the cseabreeze backend?

island-monkey, 2018-03-26:
Got it running with the cseabreeze backend. Thanks for your time and help !

ap--, 2019-09-11:
Closing :+1:

closed 2019-09-11 by ap--