ap--/python-seabreeze #56: Installation failure

cwhanley, 2018-02-14:
Thank you for all the effort putting this stuff together, but I am running into a problem with installation of python SeaBreeze.  I have tried to install with anaconda (and by hand) on a windows 7 64bit computer, python 3.6. I have also tried anaconda install on Linux CentOS 7 64bit, python 2.7.

In all cases I get warnings that the backend driver can not be found, or that a lib could not be found. Anaconda said all requirements have been meet.  Do I need to compile/install the Ocean Optics C/C++ SeaBreeze driver separately?

I have tried to follow all instructions on the GitHub project site, but no joy.  Any suggestions on other resources I should review?

Thanks for your time.

Clark
```

ap--, 2018-02-14:
Hey Clark,

Please post the exact error message you're getting under windows and/or centos.

I'd try using anaconda python 3.5 https://docs.anaconda.com/anaconda/faq#id3
because I am not (yet) providing packages for 3.6, and if I have to guess, I believe this could be your issue.

But very soon (I believe within the next month), I'll compile new packages for 2.7 and 3.6 for all operating systems and architectures (Just very busy at the moment).

Let me know if I can be of further help.
-Andreas

ap--, 2018-02-14:
Ignore what I just said. I'm already providing 3.6 packages...

Post the error messages and we'll try to get it running on your machine :)

cwhanley, 2018-02-16:
OK,  it looks like it is just me having a problem.  8-)

Or you fixed it in the release for python 3.6, in any case installed in anaconda and ran in anaconda, but still does not run if I use the normal python 3.6 that was previously installed on my computer.

Thanks for your efforts, I seem to have been asking the wrong question. My goal is to be able to produce an application that can just be run by untrained (non-programmer) people in order to get an output from the Ocean Optics unit.

You software works, within the anaconda environment it was installed in (the one anaconda prompt window was open the whole time). I suppose what I need to ask at this point is how to get things running in normal python, or how to run this stuff in anaconda?  I am a programmer, but new to python and up until now had never heard of anaconda.  Pointer to on-line recourses would be great.

Full description of what I did, both to get it working in anaconda and then failed to run in python is shown below (user names, etc changed in order to protect the guilty):

Computer is Windows-7 64bit

Install Anaconda

                for all users

                in C:\ProgramData\Anaconda3

                using Anaconda3-5.0.1-Windows-x86_64.exe

                Register as system python 3.6

                No errors or problems reported.

Set up .condarc file for authenticating proxy

                location C:\users\[username]\.condarc

                contents

channels:

  - defaults

# ssl_verify: true

show_channel_urls: True

allow_other_channels: True

proxy_servers:

    http: http://[user]:[password]@[proxy]:[port]

    https: http://[user]:[password]@[proxy]:[port]

ssl_verify: False

Rebooted system.

Started anaconda navigator and it did an update to versions 1.6.12

Selected Anaconda prompt and did the install according to the github instructions

                conda install -c poehlmann python-seabreeze

                Install of python-seabreeze 0.6.0-py36_0 appears to have worked.

                Got a no permissions error on C:\ProgramData\Anaconda3 for updates on other packages

                Used File explorer to grant permissions to all users

                Reran the coda command in order to apply updates.

Removed existing drivers from my USB2000+ in device manager. Then reinstalled from the recommended download link on the github page.

Did some modifications on the demo file from the github page (ran in the anaconda prompt window) and everything seems to be working.

Source:

import seabreeze.spectrometers as sb

devices = sb.list_devices()

print (devices)

spec = sb.Spectrometer(devices[0])

spec.integration_time_micros(12000)

spec.wavelengths()

print (spec.intensities())

Closed the anaconda prompt window and attempted to run the same sample program from the python IDLE 3.6 application (Preexisting on the computer) and received the following error.

Traceback (most recent call last):

  File "C:\Users\[username]\AppData\Local\Programs\Python\Python36\lib\site-packages\seabreeze\backends.py", line 30, in _use_cseabreeze

    import seabreeze.cseabreeze as sbb

  File "C:\Users\[username]\AppData\Local\Programs\Python\Python36\lib\site-packages\seabreeze\cseabreeze\__init__.py", line 5, in <module>

    from .wrapper import (SeaBreezeError,

ModuleNotFoundError: No module named 'seabreeze.cseabreeze.wrapper'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "C:\Users\[username]\Downloads\SeaBreeze\first_test.py", line 1, in <module>

    import seabreeze.spectrometers as sb

  File "C:\Users\[username]\AppData\Local\Programs\Python\Python36\lib\site-packages\seabreeze\spectrometers.py", line 4, in <module>

    lib = seabreeze.backends.get_backend()

  File "C:\Users\[username]\AppData\Local\Programs\Python\Python36\lib\site-packages\seabreeze\backends.py", line 15, in get_backend

    b = _use_cseabreeze()

  File "C:\Users\[username]\AppData\Local\Programs\Python\Python36\lib\site-packages\seabreeze\backends.py", line 32, in _use_cseabreeze

    warnings.warn("Can't load seabreeze c library wrapper, because:\n'%s'\n" % err.message )

AttributeError: 'ModuleNotFoundError' object has no attribute 'message'

ap--, 2019-09-11:
Conda packages and Pypi wheels are provided for v1.0.0 :tada: 

https://python-seabreeze.readthedocs.io/en/latest/install.html

closed 2019-09-11 by ap--