ap--/python-seabreeze #57: Limited functionality for context manager

dave-ford, 2018-04-10:
### spectrometer and system information ###

- model: all
- operating system: Debian
- python version: 3.5
- python-seabreeze version: current master
- installed-via: instructions in this repo for ARM

### current problem ###

The context manager for the Spectrometer object (included below) will close the connection on `__exit__` but it doesn't re-open it on `__enter__`. Ideally, I would like to use this in a manner like this:

```
spec =  sb.Spectrometer.from_serial_number('xyz')
with spec as s:
    intensities = s.intensities()
    wavelengths = s.wavelengths()
```

And these are the context managers now:

```
    def __enter__(self):
        return self

    def __exit__(self, type, value, traceback):
        self.close()
```

And, for comparison, this is what pyserial does for a Serial object (see SerialBase at https://github.com/pyserial/pyserial/blob/master/serial/serialutil.py):

```
    def __enter__(self):
        if not self.is_open:
            self.open()
        return self

    def __exit__(self, *args, **kwargs):
        self.close()
```

Of course, I could create a new Spectrometer instance each time I use it, but that seems annoying and it also wastes the effort to find the Device object. I think it would be nice to have a simple `Spectrometer.open` method that doesn't take any arguments, and use it in the `__enter__` method.

So, do others agree that it would be useful to have the feature of allowing the context manager to open and close the connection?

Happy to take a stab at it if others agree.

ap--, 2018-04-10:
Hi Dave,

thanks for contributing!

The reason I added the context managers like this, was just so I could use a with statement instead of a try...finally block around the global measurement logic and guarantee that the spectrometer is closed. I am not entirely sure if opening and closing the spectrometer connection for every measurement will work, because Spectrometer._open_device() calls lib.device_open(self._dev) which does some spectrometer initialization. (I do believe the initialization is a no-op in cseabreeze for most spectrometers, but I think not for all) 

(In the pyseabreeze case it resets the usb connection, which might not be desirable while you run your experiment)

So what I am actually saying is:
If you go ahead with implementing the change and submitting a pull request, we should wait with merging it into master until we can confirm correct operation for a few different spectrometer models and operating systems.

I can test a USB2000+ and a Spark under Linux, Windows and OSX with both backends.

Thanks,
-Andreas

ap--, 2020-08-10:
I'm closing this issue since automatic closing of spectrometers is quite a bit improved and the contextmanager fuinctionality will be removed with v2.0

closed 2020-08-10 by ap--