ap--/python-seabreeze #58: Implementation for the NIR256 2.5

anffray, 2018-06-29:
Hello !
### spectrometer and system information ###

- model: NIR256 2.5
- operating system: Windows 8.1 64bits
- python version: 2.7
- python-seabreeze version: 0.6.0
- installed-via: anaconda

### current problem ###

I have a python program which uses the data transmitted by a spectrometer. Everything is working well with the HR4000 but I would like to make my program work with the NIR256 2.5 (which is older). It is also a device from Ocean Optics so it should work easily...

### steps to reproduce ###

When I execute the simple code found out here : 
```
import seabreeze
seabreeze.use("pyseabreeze")
#seabreeze.use("cseabreeze")
import seabreeze.spectrometers as sb
devices = sb.list_devices()
print(devices)
```
With the Hr4000 no problem, it is detected and everything is good but with the nir256 it just print [ ] : so nothing indeed...  

So I found out that it didn't work because this spectrometer wasn't defined in the files so I tried to add it myself where I could find the other spectros mentionned, so in C:\ProgramData\Anaconda2\Lib\site-packages\seabreeze\pyseabreeze\interfaces I changed these files :

-  __init__.py , where I tried to create a class using the datasheet such as : 
```
class NIR256(SpectrometerFeatureNIR256,
             WavelengthCoefficientsEEPromFeature,
             NonlinearityCoefficientsEEPromFeature,
             EEPromFeature,
             NoShutterFeature,
             NoTecFeature,
             NotImplementedWrapper):
    _ENDPOINT_MAP = EndPoints['NIR256']
    _PIXELS = 256  # FIXME
    _RAW_SPECTRUM_LEN = (256 * 2) + 1
    _INTEGRATION_TIME_MIN = 1000
    _INTEGRATION_TIME_MAX = 655350000
    _INTEGRATION_TIME_BASE = 1000
    _MAX_PIXEL_VALUE = 16383
```
 but I am not sure for the values...
- defines.py , with the productID 0x1010 but I didn't know the list range so I put [ ]. I also chose endpoints : _EML2K and TriggerModes :     
```
'NIR256'	   : { 
       'FREE_RUNNING' : 0,
       'SOFTWARE'     : 1,
        },
```
- spectrometer.py , I put 
```
class SpectrometerFeatureNIR256(SpectrometerFeatureOOI2K):
    pass
```
I also found out about this driver thing and with the same method I created an .inf file in the folder C:\Program Files\Ocean Optics\SeaBreeze\Drivers by copying on the other .inf of the spectrometers.
In the Ocean Optics folder I have seabreeze and OmniDriverr

I tried to use both pyseabreeze and cseabreeze but nothing  works ! (in the cseabreeze folder there is not much beside a __init__.py file) I am a bit confused between OmniDriver, pyseabreeze, cseabreeze, pyusb or usb.core...

I hope I was clear enough and that you will be able to help me...
I an very new to python and seabreeze and this driver/usb-connection is driving me crazy trying to make it work for the nir256... 
I don't really know what to do now... The device is detected by my PC because there is the  icon in the notification area and it is in the device manager...

If you have any questions do not hesitate, thanks !!

PS : for the values I used this datasheet : https://oceanoptics.com/wp-content/uploads/OEM-Data-Sheet-NIR.pdf

ap--, 2018-06-30:
So according to the datasheet the NIR256 has the product ID 0x1010 and if I understand you correctly you created an inf file for the NIR256 so that it uses the pyusb compatible windows-driver.

So if you did this the correctly you should be able to run:

```python
import usb.core
list(usb.core.find(idVendor=0x2457, idProduct=0x1010, find_all=True))
```

and it should find the Device.

**If it finds it:**
make a pull request with your changes to pyseabreeze and I'll have a look at your code changes.
**If it doesn't:**
search for tutorials online on how to use pyusb with a usb device on windows, and try to install the correct driver for your spectrometer.

Good luck

anffray, 2018-07-02:
Hi !
The device is now well detected by the pc as the code lines you gave me print this on the spyder console : 
`[<DEVICE ID 2457:1010 on Bus 000 Address 001>]`

But the seabreeze's list_devices() is still empty unfortunately...:/
I will make a pull request, I hope you will find the errors...

Thanks !

ap--, 2018-07-03:
hmm...
can you report the output of:
```python
import seabreeze.pyseabreeze.interfaces.defines as sbdef
print sbdef.ProductIds
```

anffray, 2018-07-04:
This is what I get : [4160, 8192, 4098, 4164, 4166, 4134, 4136, 4106, 4140, 4130, 16384, 4112, 4114, 20480, 4116, 4118, 4120, 16388, 4138, 4126]

ap--, 2018-07-19:
Hey @anffray,
sorry for the late reply. I can't really see an issue with your code (although I might miss something. It's hard to debug without the hardware) 

one more thing you could try is:
```python
import usb.core
from seabreeze.pyseabreeze.interfaces import NIR256
import numpy as np

dev, = list(usb.core.find(idVendor=0x2457, idProduct=0x1010, find_all=True))
s = NIR256()
s.open_device(dev)
N = s.get_formatted_spectrum_length()
arr = np.zeros((N,))
s.get_formatted_spectrum(arr)
print arr
```

anffray, 2018-07-26:
Thank you !

So I tried this and with my NIR I got : 
```
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda2\lib\site-packages\seabreeze\pyseabreeze\interfaces\common.py", line 14, in decorated_func
    return func(*args, **kwargs)
  File "C:\ProgramData\Anaconda2\lib\site-packages\seabreeze\pyseabreeze\interfaces\spectrometer.py", line 139, in get_formatted_spectrum
    out[:] = ret * self._NORMALIZATION_VALUE
AttributeError: 'NIR256' object has no attribute '_NORMALIZATION_VALUE'
```
With another try I got this and I have the same SeabreezeError with a HR4000 : 
 ```
Traceback (most recent call last):

 File "<ipython-input-2-4c110a6164d0>", line 1, in <module>
    runfile('C:/Users/AAA/Desktop/testscript.py', wdir='C:/Users/AAA/Desktop')

  File "C:\ProgramData\Anaconda2\lib\site-packages\spyder\utils\site\sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)

  File "C:\ProgramData\Anaconda2\lib\site-packages\spyder\utils\site\sitecustomize.py", line 87, in execfile
    exec(compile(scripttext, filename, 'exec'), glob, loc)

  File "C:/Users/AAA/Desktop/testscript.py", line 19, in <module>
    s.get_formatted_spectrum(arr)

  File "C:\ProgramData\Anaconda2\lib\site-packages\seabreeze\pyseabreeze\interfaces\common.py", line 23, in decorated_func
    raise SeaBreezeError(msg)

SeaBreezeError: Error while reading raw spectrum
```

I don't really know where these errors come from...

ap--, 2018-07-26:
Sorry, my mistake:
```
import usb.core
from seabreeze.pyseabreeze.interfaces import NIR256
import numpy as np

dev, = list(usb.core.find(idVendor=0x2457, idProduct=0x1010, find_all=True))
s = NIR256()
s.open_device(dev)  

# < ADDED THE FOLLOWING LINES
s.usb_send(struct.pack('<B', 0x01))
time.sleep(0.1)
s._initialize_common()
# />

N = s.get_formatted_spectrum_length()
arr = np.zeros((N,))
s.get_formatted_spectrum(arr)
print arr
```

anffray, 2018-07-26:
So it gave this to me : 
```
[    0.   912.   514.   722.  1294.   278.  1394.  3941.  1367.  1373.
   694.  1298.   314.  2826.  2663.  1728.  1446.   541.  3092.  1747.
  3734.  1239.  2302.    39.  3568.   387.    27.  3894.  1794.   267.
  1758.  3912.  2184.  2085.  3932.  3620.  3648.   906.  2784.  3771.
  1579.  3210.  2536.  3468.  1718.  3094.  1653.   586.  2147.  2120.
   426.  1718.   393.  4054.  1166.   702.   671.  3955.    94.   592.
  1566.   691.  1162.  2604.  3366.   727.  1231.  1730.  3918.  1658.
   340.   350.  1998.   266.  3234.   472.  3748.   132.  2589.   884.
  3334.  2798.  1059.  3424.   514.   491.   795.  2895.  1372.  2338.
  1516.   496.  2174.  3714.  1250.  3989.  2547.   873.  2766.  1470.
  3237.  1430.  3870.   246.  2984.  2959.  4052.  1994.  2315.  3866.
  2278.  2172.  3796.  3195.   278.  1888.  3990.  3200.  1287.   660.
  3877.   502.  1336.   133.   887.  3316.  1643.  3402.   754.  4067.
  3177.  3970.   231.  2286.   629.   914.  3066.   834.  2642.  2806.
   368.  3588.  3902.  1606.  1630.  2831.  1665.  2031.  3770.  3244.
   916.  1180.  1284.  3114.   512.  2386.  2246.  2694.  1244.  2819.
  2214.  2442.  4091.  3848.  1002.   495.   681.  2404.    36.  2116.
     6.  2363.  2034.  3678.  3126.  1482.  3183.  2142.  3646.   422.
  2084.    95.   442.  1662.  2154.  3870.   836.  3255.   910.  3866.
  1972.  2176.  1429.  2562.  3596.  2902.  1998.  3360.  2494.  1130.
  2258.   775.  2726.  1075.   922.  1270.  3619.   130.  3868.    62.
  2447.  2830.  2719.  3909.    62.  1582.  1011.  1258.  2091.  2107.
   360.  3579.  2564.  3761.  2290.  3426.   618.   788.  4070.  2082.
  3978.   705.  1371.  1630.  3290.  1977.  2319.  2386.   218.   116.
  3852.  3882.  3423.  2779.   622.  2714.  4095.  1799.  3891.  2074.
    94.   748.  3266.  2743.  3694.  3247.]
```

ap--, 2018-07-28:
Well looks like you're getting closer to a working implementation for your spectrometer model :+1: 

If this does not look like a spectrum, when you plot it, try using `get_unformatted_spectrum_length` and `get_unformatted_spectrum` record one spectrum with the light blocked, and one with the spectrometer saturated. Then you can try to reverse engineer the bitpattern for this spectrometer.

Look at the different implementations of `get_formatted_spectrum` in pyseabreeze to get a feeling for how to reconstruct the spectra. Most of them are just a bit of byte shuffling to reconstruct the words per pixel and then maybe an XOR for some of the higher bits.

And look at the spectrometer class in seabreeze to see how to use the lower level interface directly.
Here's the example for how to query the wavelength array (which I would try next, in your case) https://github.com/ap--/python-seabreeze/blob/master/seabreeze/spectrometers.py#L112-L118

Keep up the good work :medal_sports: