ap--/python-seabreeze #64: Can't read PCB temperature on a Flame spectrometer

eduardosprp, 2019-02-20:
Hi. Sorry if this text is badly worded or formatted, I created this account just so I could solve this issue.

### Spectrometer and system information: ###

- Flame (recognized as USB2000PLUS by pyseabreeze
- OS: Raspbian
- python version: 3.6.1
- python-seabreeze version: current master
- installed-via: conda
- System: Raspberry Pi 3 model B

### Current problem ###

I know that the Flame spectrometer is able to read its PCB temperature, as stated in its user guide and verified by personal experience with a previous project, in which a PC running Windows and official software was used to control the spectrometer. However, whenever I try using tec_get_temperature_C(), I get 'No such feature on device'.

### Steps to reproduce ###

Here is the minimal code example, along with the error log:

import seabreeze.spectrometers as sb
spec = sb.Spectrometer(sb.list_devices()[0])
spec.tec_get_temperature_C()

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/berryconda3/lib/python3.6/site-packages/seabreeze/spectrometers.py", line 227, in tec_get_temperature_C
    return lib.tec_read_temperature_degrees_C(self._dev, self._fidte)
  File "seabreeze/cseabreeze/wrapper.pyx", line 509, in seabreeze.cseabreeze.wrapper.tec_read_temperature_degrees_C (./seabreeze/cseabreeze/wrapper.c:12136)
seabreeze.cseabreeze.wrapper.SeaBreezeError: b'Error: No such feature on device'

ap--, 2019-03-03:
Hey eduardo,

As the error message states: your spectrometer does have `No such feature`, which means the TEC temperature control feature.

If you want to read the PCB  temperature on your flameX you need to send a custom command to the spectrometer, which is currently not exposed in the **cseabreeze** backend, but you should be able to do it with the **pyseabreeze** backend.

```diff
-! BUT BE AWARE:                                                                       !-
-! this is a reading of the PCB temperature. This is most likely not what you actually !-
-! want, which I guess is the sample temperature, or the environment temperature.      !-
```

The following code is untested:
```python
import seabreeze
seabreeze.use('pyseabreeze')
import seabreeze.spectrometers as sb
import struct

# open first spectrometer
spec = sb.Spectrometer.from_serial_number()

# query PCB temperature
# https://oceanoptics.com/wp-content/uploads/FlameIO.pdf, page 79f
spec._dev.interface.usb_send(struct.pack('<B', 0x6C))
raw_bytes_response = spec._dev.interface.usb_read_lowspeed(timeout=1000)
temp = 0.003906 * struct.unpack("<h", raw_bytes_response[1:3])  # should be 0x08 LSB MSB 

print "PCB temperature", temp
```

Good luck.

eduardosprp, 2019-03-03:
Thanks, I'll test this as soon as I can, but I think it'd be useful to mention that I've already tried this solution https://github.com/ap--/python-seabreeze/issues/37 with a STS spectrometer, and it didn't work.
The error log mentioned that there were unsupported operand types for "%": 'int' and 'str', (or 'int' and 'bytes', if I'm not mistaken). I then tried running this on python 2.7, and tinkering with the code, but neither got anywhere.

ap--, 2019-03-03:
> I've already tried this solution https://github.com/ap--/python-seabreeze/issues/37 with a STS spectrometer, and it didn't work. The error log mentioned that there were unsupported operand types for [...]

https://github.com/ap--/python-seabreeze/issues/37#issuecomment-284399536
Seems to indicate that it does work.

Can't help without a traceback. Your description sounds like it's just a trivial error.

eduardosprp, 2019-03-03:
Alright then. I'll see if I can test this by tomorrow and I'll post the error in full here, if there are any. Thank you for your patience.

eduardosprp, 2019-03-04:
Wow, it actually worked, thank you so much! Still, the code you submitted returns an error due to the fact that the variable "temp" is actually an array, and multiplication of arrays can only be done if the multiplier is an integer. The following alteration fixes this very minor problem:
`#temp is actually interpreted as a list of numbers, even though it consists of a single value`
`#doing as such fixes the problem:`
`print("PCB temperature", temp[0]*0.003906)`

closed 2019-03-04 by eduardosprp