ap--/python-seabreeze #65: No Devices Found, Possible issue with driver

michelleanselmo98, 2019-06-14:
### spectrometer and system information ###

- model: OceanOptics HR4000
- operating system: Windows 10, 64 bit
- python version: Python 3.7.3 64 bit
- python-seabreeze version: current master
- installed-via: conda

### current problem ###

I followed instructions exactly to install the spectrometer with the linked driver files for window and everything worked until I tried running the code below. It doesn't seem like my spectrometer is recognized?

### steps to reproduce ###

import usb.core
usb.core.find()

import seabreeze
seabreeze.use("pyseabreeze")
import seabreeze.spectrometers as sb

devices = sb.list_devices()
print(devices)

Exit Code:

[]

Process finished with exit code 0

I am a beginner so there may be a small problem that I am not aware of.

gabriel-cruz, 2019-06-17:
I have the same problem, but I am using a Raspberry Pi

ap--, 2019-11-05:
Closing because of duplicate issue #72
But the HR4000 does work with v1.0.0 and cseabreeze backend.

closed 2019-11-05 by ap--