ap--/python-seabreeze #66: Udev rules does not work on Raspberry Pi 3

gabriel-cruz, 2019-06-17:
### spectrometer and system information ###

- model: USB2000+
- operating system: Raspbian newer version
- python version: Python 2.7.13
- python-seabreeze version: current master
- installed-via: conda (Berryconda)

### current problem ###

Installation of python-seabreeze is fine, but when i connect the spectrometer in rpi in the `lsusb` command does not show anything

### steps to reproduce ###

_Step by step instructions to reproduce the error. The more detailed the better, but please use some common sense:_

1. I followed all the steps of the [tutorial](https://github.com/ap--/python-seabreeze) and I installed the pyseabreeze correctly in the rpi
2. In python running
`import seabreeze.spectrometers as sb
sb.list_devices()`

returns

[]

I did not find a solution anywhere, please help

ap--, 2019-07-04:
hi @gabriel-cruz 

if `lsusb` doesn't show it, something else must be wrong. Does this spectrometer work at all?

gabriel-cruz, 2019-07-08:
I tried connecting on windows but I also can not connect to the driver. Could it be a problem with the spectrometer?

ap--, 2019-07-08:
Sounds likely. I would first try getting it to work with the original oceanoptics software.

ap--, 2019-11-05:
Closing for now. Please reopen if this is not resolved :rainbow:

closed 2019-11-05 by ap--