ap--/python-seabreeze #68: Need help implementing new methods

llmorreale, 2019-07-23:
### spectrometer and system information ###
- model: QEPRO
- operating system:  Tried on both Windows 10 & Ubuntu 16.4
- python version: 2.7.16
- python-seabreeze version: current mastter
- installed-via: setup.py

### current problem ###
I am trying to implement new methods (specifically having to do with managing the buffer on the QEPRO).  I have no problem getting seabreeze to run following the conda install instructions. However, when I try to build a new version of the python-seabreeze package locally (from cloned github source), I cannot, because it cannot find the correct usb library.

I have attempted this on both windows and ubuntu, and run into the same problem in both (even though in the code in setup.py you can see that they rely on different usb libraries). 
On windows, i have WinUSB installed and on ubuntu i have the libusb package.

Any suggestions, or help, from anyone who has implemented new methods would be greatly appreciated.

closed 2019-07-23 by llmorreale

ap--, 2019-07-24:
Did you follow the instructions [here](https://github.com/ap--/python-seabreeze/blob/master/docs/LINUX_INSTALL.md) ?

If you post the full error message it'll be easier to help :)

llmorreale, 2019-07-24:
Thanks for responding!! I was able to get it working on Linux. The issue was that I had the non-dev version of libusb installed.  And thank you very much for this package - it is a huge help to our lab!