ap--/python-seabreeze #69: HR4000 returning data transfer error

Ganar-lab, 2019-10-14:
### spectrometer and system information ###

- model: HR4000
- operating system: Windows 10 64bit
- python version: Python 3.7.4
- python-seabreeze version: current master
- installed-via: pypi  (pip install seabreeze)

### current problem ###

Hi, I am just begining to use python-seabreeze to monitor my spectrometer HR4000. I followed the installation instructions, all with success. First it was not possible to install all the drivers through seabreeze_os_setup (because windows blocked all the unsigned drivers), but using a workaround, I installed them all.
Now, I detect the spectrometer, but I get an error when trying to get the intensities, See bellow.
Could you help ?

Note: the spectrometer is working as it works on an old computer using LabView installation (we want to use python on the new computer).

### minimal code example and error (very helpful if available) ###

```python
import seabreeze
from seabreeze.spectrometers import list_devices, Spectrometer

spec = Spectrometer.from_first_available()
print(spec)
print(list_devices())
wavelengths = spec.wavelengths()
print(wavelengths)
intensities = spec.intensities(correct_dark_counts=False, correct_nonlinearity=False)
print(intensities)
```
output:
```
<Spectrometer HR4000:HR4C2575>
[<SeaBreezeDevice HR4000:HR4C2575>]
[759.8951654  759.92323596 759.95130346 ... 841.90731038 841.92423657
 841.9411597 ]
Traceback (most recent call last):
  File "spectrometer.py", line 17, in <module>
    intensities = spec.intensities(correct_dark_counts=False, correct_nonlinearity=False)
  File "C:\Users\User\Anaconda3\envs\spec\lib\site-packages\seabreeze\spectrometers.py", line 180, in intensities
    out = self._dev.f.spectrometer.get_intensities()
  File "src\seabreeze\cseabreeze\c_seabreeze_wrapper.pyx", line 799, in seabreeze.cseabreeze._wrapper.SeaBreezeSpectrometerFeature.get_intensities
seabreeze.cseabreeze._wrapper.SeaBreezeError: Error: Data transfer error
```

ap--, 2019-10-14:
Hmm, that's definitely unexpected.
Please clone the repository and try running the pytests in your conda environment.

https://python-seabreeze.readthedocs.io/en/latest/contributing.html#testing-spectrometer-models

And post the output here.

Also if you could make a pull request updating the documentation with the workaround you had to do to install the unsigned drivers, that would be great :)

Best,
-Andreas

Ganar-lab, 2019-10-15:
I've run the tests.

**First tests:**
[pytest_output1.txt](https://github.com/ap--/python-seabreeze/files/3729922/pytest_output1.txt)
This test run rises an error:
`error: Don't know how to compile src\seabreeze\cseabreeze\c_seabreeze_wrapper.pyx`
I then run pip install cython and rerun as suggested by the setup INFO.

**Second test**
[pytest_output2.txt](https://github.com/ap--/python-seabreeze/files/3729931/pytest_output2.txt)
The final error is:
`E   usb.core.NoBackendError: No backend available`

For the workaround, I will do when I have time (even though I'm not sure whether it is necessary to do so as installing the unsigned drivers didn't change much - though I believed it was the problem).

Best.

ap--, 2019-10-15:
Strange. I think something might be wrong with your conda environment.
Can you try the following:
- create a new conda environment and activate it
- conda install seabreeze via the conda-forge channel as described in the docs and not via pip
- navigate to the cloned repository root and run `python setup.py pytest`

Best,
-Andreas

Ganar-lab, 2019-10-16:
Here the complete steps:
[pytest_output3.txt](https://github.com/ap--/python-seabreeze/files/3733183/pytest_output3.txt)

Important points: 

**First** an error occured when running seabreeze_os_setup in the new env. Fortunately, it does not make any problem as the drivers where previously installed and the installation is system wide.
error: 
`Fatal error in launcher: Unable to create process using '"c:\bld\seabreeze_1568525710274\_h_env\python.exe"  "C:\Users\User\Anaconda3\envs\spectest\Scripts\seabreeze_os_setup.exe" '`

**Next**, you can ignore the first run of python setup.py pytest as I didn't plugged in the spectrometer yet.

**Finally**, the test end up testing the intensities and got stuck. It didn't respond anymore so that I needed to close the terminal.

Any Idea ?

ap--, 2019-10-18:
Okay, seems the pyseabreeze tests get stuck on reading the spectrum...

In your conda environment, with the spectrometer attached, can you try:

```python
>>> import seabreeze                                                                                 
>>> seabreeze.use('cseabreeze')                                                                      
>>> from seabreeze.spectrometers import Spectrometer                                                 
>>> spec = Spectrometer.from_first_available()                                                       
>>> spec.intensities()                                                                               
array([1498., 1503., 1498., ..., 1495., 1500., 1504.])
>>> spec.intensities()                                                                               
array([1499., 1500., 1502., ..., 1494., 1502., 1504.])
>>> spec.intensities()                                                                               
array([1503., 1499., 1508., ..., 1500., 1506., 1501.])
```
this just tests if cseabreeze works with your spectrometer. Close the interactive session afterwards.

and then try in a new python shell:
```python
>>> import seabreeze                                                                                 
>>> seabreeze.use('pyseabreeze')                                                                      
>>> from seabreeze.spectrometers import Spectrometer                                                 
>>> spec = Spectrometer.from_first_available()                                                       
>>> spec.intensities()                                                                               
array([1498., 1503., 1498., ..., 1495., 1500., 1504.])
>>> spec.intensities()                                                                               
array([1499., 1500., 1502., ..., 1494., 1502., 1504.])
>>> spec.intensities()                                                                               
array([1503., 1499., 1508., ..., 1500., 1506., 1501.])
```
which tests the same with the pyseabreeze backend.

Ganar-lab, 2019-10-18:
Hi. So, I have tested the backends several times (see below why).
The cseabreeze backend works it seems. That is nice.
The pyseabreeze backend does not manage to access the device it seems,
Here is the sequence of tests that I made. Going from one to the next means that I closed the previous interactive session, and I have write in the begining of the next file if I have restarted the computer in between. I did so because I believed that the denied access was due to the usb still being used. But the last tests seem to show that this is not the reason,

pyseabreeze  does not work:
[backend_test_output1.txt](https://github.com/ap--/python-seabreeze/files/3744613/backend_test_output1.txt)
cseabreeze works:
[backend_test_output2.txt](https://github.com/ap--/python-seabreeze/files/3744639/backend_test_output2.txt)
pyseabreeze  does not work:
[backend_test_output3.txt](https://github.com/ap--/python-seabreeze/files/3744642/backend_test_output3.txt)
cseabreeze works, and some tests of close and open options (Data transfer error legitimate):
[backend_test_output4.txt](https://github.com/ap--/python-seabreeze/files/3744616/backend_test_output4.txt)
pyseabreeze  does not work:
[backend_test_output5.txt](https://github.com/ap--/python-seabreeze/files/3744617/backend_test_output5.txt)
cseabreeze works, repeatably, even without restart after errors from previous session with pyseabreeze:
[backend_test_output6.txt](https://github.com/ap--/python-seabreeze/files/3744654/backend_test_output6.txt)

From here, I think this is ok for me to start my project using cseabreeze. Thanks for helping me !
I will report if there is any new error.

ap--, 2019-10-19:
Interesting. And happy that it works now :)

So I think the issues are:
- the pyseabreeze backend is broken for the HR4000
- your very first conda environment had a broken seabreeze installation (I guess because you pip installed in the conda environment instead of using the conda package and who knows which compiler it used and what usb library it linked against...)

But I think we can close this issue now :+1: 

Open a new issue if there's more problems :)

closed 2019-10-19 by ap--