ap--/python-seabreeze #74: correct_nonlinearity=True gives incorrect intensities

acpo, 2019-11-12:
### spectrometer and system information ###
- model: HR4000
- operating system: Raspbian (Buster) freshly installed last week
- python version:  2.7.16 and 3.7.3 (tested in both)
- python-seabreeze version: current master
- installed-via: python setup.py install; really via pip

### current problem ###
When the correct_nonlinearity option is set true  {spec.intensities(correct_nonlinearity = True) } the data returned is incorrect.  The numerical values alternate between very small and very large, like 1e-5 to 1e8.
When set to false, the data is correct.  So there is an easy short-term fix.
However, the version of python-seabreeze from about a year ago didn't have this issue.

### steps to reproduce ###
With an operating spectrometer,
Try   ```spec.intensities(correct_nonlinearity=True)```
then look at ```spec.intensities(correct_nonlinearity=False)```

ap--, 2019-11-13:
That's unexpected.

could you post the return value of
```python
# nonlinearity coefficients
spec._nc
# and dark pixel indices
spec._dp
```

Best,
Andreas

acpo, 2019-11-13:
>>> import seabreeze.spectrometers as sb
>>> spec = sb.Spectrometer.from_serial_number()
>>> spec._nc
[0.939935, 4.48307e-05, -1.58006e-08, 3.23411e-12, -3.96006e-16, 2.8124e-20, -1.05914e-24, 1.61494e-29]

>>> spec._dp
[5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17]
>>>

ap--, 2019-11-14:
@apco
thanks! found the issue! Currently all spectrometers that support non_linearity correction should return broken spectra when non_linearity correction is enabled.

Current Code:
https://github.com/ap--/python-seabreeze/blob/59c9f6dc1787735835d0accd41170c9a31fbf7fb/src/seabreeze/spectrometers.py#L67

Should be:
https://github.com/ap--/python-seabreeze/blob/8b50ed52fad622450ea5d16d5679544881a07ea0/seabreeze/spectrometers.py#L125

### workaround for now

You should be able to do the following to get nonlinearity correction working until I release a new version:
```python
import numpy as np
from seabreeze.spectrometers import Spectrometer

spec = Spectrometer.from_first_available()
# do the following to correct the nonlinearity coefficients:
spec._nc = numpy.poly1d(spec._nc[::-1])

# now it should work correctly
spec.intensities(correct_nonlinearity=True)
```

acpo, 2019-11-15:
Thanks!  and I look forward to the new version.

commit 1bfbe160 (2019-11-16): [spec] fix nonlinearity coefficients regression

Closes #74.
closed 2019-11-16 by ap--