ap--/python-seabreeze #76: seabreeze_setup_os does not run

aoabdelatif, 2020-01-31:
### spectrometer and system information ###

- model: HR4000
- operating system: Windows 10 64bit
- python version: 3.2
- python-seabreeze version: 1.0.1
- installed-via: conda

### current problem ###

 After successfully installing sea breeze via conda, seabreeze_os_setup.exe does not run. 

Fatal error in launcher: Unable to create process (see attached)

Manually locating seabreeze_os_setup and run by administer does not work either, application immediately crashes. 

### steps to reproduce ###

1. install seabreeze via conda command prompt
2. run seabreeze_setup_os via anaconda command prompt

### minimal code example and error (very helpful if available) ###

In the anaconda command prompt

>>>c conda-forge seabreeze
>>>seabreeze_os_setup
>>>"Fatal Error..."

see attached screenshot for errors from command prompt and downloaded zip

[OceanOptics.zip](https://github.com/ap--/python-seabreeze/files/4141837/OceanOptics.zip)

ap--, 2020-02-01:
Error message in zip:
[image]

@aoabdelatif 
could you post the output of `conda list`

aoabdelatif, 2020-02-03:
outout of `conda list` (see attached txt)

[conda_list.txt](https://github.com/ap--/python-seabreeze/files/4148662/conda_list.txt)

After reading through some of the other issues, it seems that i may need to figure out the work around to install seabreeze_os_setup similar to issue #69, did this user ever get back on how to do this?

ap--, 2020-02-03:
Hmm. You could try running:
```
python -I -m seabreeze.os_setup
```

And if that doesn't work make a clean conda envorionment and try again:
```
conda create -n seabreeze-test -c conda-forge seabreeze
conda activate seabreeze-test
```

aoabdelatif, 2020-02-03:
OK, i tried these commands and i still get the "Fatal Error in launcher". (see attached)

[conda_clean.txt](https://github.com/ap--/python-seabreeze/files/4149971/conda_clean.txt)

After downloading and isntalling the 30day free trial of OceanView, everything seems to work fine. I believe ocean view was able to install the drivers i needed since Seabreeze_os_setup did not work.

I am not sure what will happen after the 30day free trial ends but i think the drivers are now on my machine and there to stay.

You can close this Issue, its all working now that I installed OceanView.

Getting the seabreeze_OS_SETUP to work sounds like more trouble than its worth. There could be a firewall on my machine that is preventing the setup to run but im not sure.

Thank you Andreas.

ap--, 2020-02-03:
Glad that it works :)
And I think I found the error. I believe it should work if you run in an admin shell with the conda environment activated: 
```
python -m seabreeze.os_setup
```

The reason it fails is that: `__file__` in your case endswith `.exe`, which is weird. because it should point to the `os_setup.py` file. This needs further investigation.

https://github.com/ap--/python-seabreeze/blob/1e4c6a5e6e4b9cc8171807335246484e9aac1af1/src/seabreeze/os_setup.py#L150-L154

But it's awesome that you found a workaround!

ap--, 2020-02-04:
NOTE to self: similar problem here: https://github.com/conda-forge/yggdrasil-feedstock/pull/4

ap--, 2020-08-23:
I believe this was fixed by:
https://github.com/conda-forge/seabreeze-feedstock/commit/a8c7d7592d145a39799b527381972c149ab3607c
I can't reproduce this with the current conda packages on windows 10.

closed 2020-08-23 by ap--