ap--/python-seabreeze #77: Driver version compatibility?

MollyInnerPlant, 2020-02-24:
### spectrometer and system information ###

- model: _FLAME
- operating system: _Windows 10
- python version: _python3.7
- python-seabreeze version: _current master
- installed-via: _ python setup.py install)_

### current problem ###

I'm having trouble getting seabreeze to recognize my device. I did have it working with downloaded drivers from Ocean Optics, but changed paths when to be compatible with a custom piece of Java software. I now cannot get it to work...I'm wondering if I need to downgrade some drivers (WinUSB from 1.5.? to 6.?.?) since they were updated. 

### troubleshooting
I did uninstall and reinstall the drivers, but no luck.

ap--, 2020-02-24:
Hi @MollyInnerPlant 

Since you manually installed via `python setup.py install` try running `python setup.py pytest` in your git checkout and post the output here.

you might have to install pytest and pytest-runner before: `pip install 'pytest-runner' 'pytest<5'`

All the best,
Andreas

ap--, 2020-06-09:
Please reopen if needed :relaxed:

closed 2020-06-09 by ap--