ap--/python-seabreeze #78: Can't find device on Linux without sudo

e-champenois, 2020-03-06:
### spectrometer and system information ###
- model: HR2000+
- operating system: Linux
- python version: 3.7.4
- python-seabreeze version: 1.0.1
- installed-via: pip3

### current problem ###
I am only able to detect my spectrometer with root privileges (starting python with sudo). If I don't use sudo, list_devices() returns an empty array and warns "usb_claim_interface() returned -1 - did you copy os-support/linux/10-oceanoptics.rules to /etc/udev/rules.d?"

import seabreeze.spectrometers as sb
devs = sb.list_devices() # This gives the warning above.
print(devs) # This prints an empty array.

ap--, 2020-03-06:
Hi @e-champenois 

This sounds like a permission problem. Did you run `seabreeze_os_setup` after pip installing?

herzig, 2020-03-09:
Hi,

I had the same issue with an STS-VIS on Rpi.
I noticed that it worked when I plugged the spectrometer in after boot (or unplug and plug back in after boot). When the device was plugged in during boot it didn't work.

Permissions when plugged in during boot:
````
ls -l /dev/sts-3
lrwxrwxrwx 1 root root 15 Mar  9 10:54 /dev/sts-3 -> bus/usb/001/004
ls -l /bus/usb/001/004
crw-rw-r-- 1 root root 189, 3 Mar  9 12:30 /dev/bus/usb/001/004
````
Permissions when plugged in after boot:
````
ls -l /dev/sts-3
lrwxrwxrwx 1 root root 15 Mar  9 12:36 /dev/sts-3 -> bus/usb/001/006
ls -l /bus/usb/001/006
crw-rw-rw- 1 root root 189, 5 Mar  9 12:36 /dev/bus/usb/001/006
````

notice that ``other`` doesn't have write permission when plugged in during boot.

I ended up solving the problem by changing ``10-oceanoptics.rules`` to set ``GROUP="plugdev"`` instead of ``MODE=0666``  (the same as it was in earlier versions e.g. https://github.com/ap--/python-seabreeze/blob/python-seabreeze-v0.6.0/misc/10-oceanoptics.rules). This works also when the device is plugged in during boot. 

``
crw-rw-r-- 1 root plugdev 189, 3 Mar  9 13:34 /dev/bus/usb/001/004
``
I don't know anything about ``udev`` and I have no idea _why_ setting ``MODE`` does not work, but maybe this can work for you too.

@ap-- what is the reason to use set permissions to 0666 instead of using the plugdev group? However I think both should work so maybe this is more of an ``udev`` on rpi issue?

### spectrometer and system information ###
- model: STS-VIS
- operating system: Linux raspberrypi 4.19.75-v7+
- python version: 3.7.3
- python-seabreeze version: 1.0.1
- installed-via: pip3 (piwheels)

ap--, 2020-03-11:
Hi @herzig

Thanks for investigating the problem and posting the solution!
The switch from using the `plugdev` group to just setting permissions to rw by all happened because `plugdev` is a debian/ubuntu specific group and not available by default on other distributions.

Getting the udev rules to work correctly everywhere is a bit of a nightmare.
Especially since there's slightly different behavior between versions of udev and systemd.

What distribution are you running on your raspberrypi?
If it's raspbian, can you post the output of:
`uname -a` and `lsb_release -a`

The fact that the permissions are set correctly when you plug the spectrometer after boot, makes me believe that it's just another bug that needs a workaround...

herzig, 2020-03-17:
Hi, I'm running Raspbian Buster (Linux raspberrypi 4.19.75-v7+).
Unfortunately I handed over the Rpi and spectrometer to someone else, and I'm no longer working on the project (fixing this issue was my last task on this). I can probably not help with any more tests.

It definitely seems weird that permissions are not set correctly by udev.

ap--, 2020-03-17:
Thanks! :+1: That's all the information I need. Just need to find a sd card, then I'll try to reproduce :)

ap--, 2020-03-29:
Okay, so this is conflicting with:
`/lib/udev/rules.d/50-udev-default.rules`

```
...
ACTION!="add", GOTO="default_end"
...
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", MODE="0664"
```

This should hit all debians I think...

Easy fix is:
```
sudo mv /etc/udev/rules.d/10-oceanoptics.rules /etc/udev/rules.d/51-oceanoptics.rules
sudo udevadm control --reload-rules
```

But let's see if I can fix this without having to change the filename.

commit 0d55567a (2020-03-29): [udev] do not allow overriding MODE

Spectrometers connected before boot should now be writable by all.

It's very likely that this bug is not limited to raspbian.
If you're on linux running `seabreeze_os_setup` will download and
install the new rules.

Close #78.
closed 2020-03-29 by ap--