ap--/python-seabreeze #79: Functionality to write eeprom data.

evading, 2020-04-01:
Hi!
I was wondering about EEPROM write functionality. More specifically to write wavelength calibration polynomial coefficients.

I can't find any way to write EEPROM anything so I was wondering if there is any specific reason.

Would you be open to a PR adding that kind of functionality?

ap--, 2020-04-01:
Hi @evading 

The reason I didn't expose eeprom writing so far, is that I am worried that it might be possible to break a spectrometer when writing something incorrect to the eeprom. And I can only test on two different models (with the firmware versions those models are currently running).

I think the only benefit that eeprom writing would currently have is exactly the functionality you want: writing non-linearity coefficients to the eeprom, so that you don't have to hardcode them in your program. But I don't know if it's worth exposing functionality that could potentially (hypothetically) break something.

With that being said, I'm happy to accept pull requests, but this functionality needs to be thoroughly tested and safeguarded. (So it would need to be locked to tested models, and only allow writing to the nonlinearity coefficient eeprom slot)

Let me know if you need additional info, or if you have any questions! I'd be happy to help :)

Cheers,
Andreas

evading, 2020-04-03:
Ok, thanks. We'll see what path we choose and if we end up implementing the coefficient writing I'll send a PR.

BR,
Einar