ap--/python-seabreeze #80: Reading temperature from the spectrometer temperature sensor

RadhaKrishnanme, 2020-04-14:
### spectrometer and system information ###

- model: _STS UV spectrometer S11908_
- operating system: _Debian 9_
- python version: _3.5_
- python-seabreeze version: 0.6.0
- installed-via: python setup.py install

### current problem ###

I would like to read the temperature of the ambient via the spectrometer temperature sensor. But it results in the following error:

seabreeze.cseabreeze.wrapper.SeaBreezeError: b'Error: No such feature on device'

### steps to reproduce ###

>>> import seabreeze.spectrometers as sb
>>> spec=sb.Spectrometer.from_serial_number()
>>> spec.tec_get_temperature_C()

### minimal code example and error (very helpful if available) ###

```python
>>> import seabreeze.spectrometers as sb
>>> spec=sb.Spectrometer.from_serial_number()
>>> spec
<Spectrometer STS:S11908>
>>> spec.tec_get_temperature_C()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/seabreeze/spectrometers.py", line 227, in tec_get_temperature_C
    return lib.tec_read_temperature_degrees_C(self._dev, self._fidte)
  File "seabreeze/cseabreeze/wrapper.pyx", line 509, in seabreeze.cseabreeze.wrapper.tec_read_temperature_degrees_C (./seabreeze/cseabreeze/wrapper.c:12153)
seabreeze.cseabreeze.wrapper.SeaBreezeError: b'Error: No such feature on device'
```

ap--, 2020-04-14:
Hi @RadhaKrishnanme,

the STS does not support the tec feature.
If you update to the newest version of python-seabreeze, you can measure the temperature of your STS as described here:

https://github.com/ap--/python-seabreeze/issues/37#issuecomment-530538238

Let me know if it works or if I can help you, and please close this issue if the instructions above resolve your issue :blush: 

Cheers,
Andreas

RadhaKrishnanme, 2020-04-15:
Hi Andreas,
I installed python-seabreeze 1.0.1 as per the instructions and executed the commands as mentioned in issue #37. I received error while reading all the temperatures. But reading the individual temperature was fine. Attached is the code for your reference.
```
>>> from seabreeze.spectrometers import Spectrometer
>>> spec = Spectrometer.from_first_available()
>>> print(spec)
<Spectrometer STS:S12090>
>>> spec.f.temperature.temperature_get_all()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "src/seabreeze/cseabreeze/c_seabreeze_wrapper.pyx", line 2875, in seabreeze.cseabreeze._wrapper.SeaBreezeTemperatureFeature.temperature_get_all
TypeError: 'int' object is not iterable

>>> a = spec.f.temperature.read_temperature(0)
>>> print (a)
39
>>> a = spec.f.temperature.read_temperature(1)
>>> print (a)
-32
>>> a = spec.f.temperature.read_temperature(2)
>>> print (a)
65

```

ap--, 2020-04-15:
Interesting!
you found two bugs :smile: 

### this needs to cast to float
https://github.com/ap--/python-seabreeze/blob/5adee55e0fa83126e7fd264291eef7b8a9c87413/src/seabreeze/cseabreeze/c_seabreeze_wrapper.pyx#L2848

### this needs to return the list of temperatures
https://github.com/ap--/python-seabreeze/blob/5adee55e0fa83126e7fd264291eef7b8a9c87413/src/seabreeze/cseabreeze/c_seabreeze_wrapper.pyx#L2875

I'll push a fix and release a new version soon :+1: 

Thanks for the detailed bug reports :tada: :medal_sports: 

Cheers,
Andreas

RadhaKrishnanme, 2020-04-16:
Thanks for the info. Do you want me to close this issue?

commit 7fe302be (2020-05-04): [csb] SeaBreezeTemperatureFeature return temperature as float

Close #80.
closed 2020-05-04 by ap--