ap--/python-seabreeze #81: pyseabreeze features implementation

moritzj29, 2020-04-28:
### spectrometer and system information ###

- model: USB650
- operating system: Windows 10
- python version: 3.7.6
- python-seabreeze version: 1.0.2
- installed-via: conda, pyusb installed via pip (current git master)

### current problem ###

First of all, thanks for this comprehensive library! Unfortunately, my spectrometer is only a USB650, so the cseabreeze backend is not supported. I got it running using the pyseabreeze backend, which is fine. Has anybody used the advanced functionalities via the pyseabreeze backend? Especially, I need to use the integrated light source of the device. I see that it should be possible with cseabreeze, but with pyseabreeze it is not implented yet, right? At least this is what I read from `seabreeze.pyseabreeze.SeaBreezeLightSourceFeature`

If anybody could give me some hints on how to get started, I would love to try implementing this. Since it is basically just an on/off switch, I hope this is feasible for me :D 

As a side note, since I found no hint on this yet:
At first I experienced problems using the pyseabreeze backend. The `list_devices()` call resulted in an execption `OSError: exception: access violation writing ...`. Some googling let me to this issue of the pyusb library https://github.com/pyusb/pyusb/issues/203, which is already fixed. So I used the current master from github instead of the conda deployed pyusb. This works fine now. Just in case anybody else is stumbling upon that...

ap--, 2020-04-28:
Hi @moritzj29 

Nice that you have the USB650 running smoothly! Last time someone was reporting bugs I assumed that support for it is broken.

The pyusb fix is great too! pyusb was without a maintainer for quiet a while and it seems as if development has started again recently! As soon as they release a new version to pypi, I can update the version requirements. In the meantime installing directly via the current pyusb master is the best option.

And regarding you question about implementing the LightSourceFeature: You would need to find a datasheet with the usb protocol for the USB650. Since we're currently assuming it's just a rebranded USB2000 a datasheet for that will do too.
In there you'll find the communication protocol for the toggling the lightsource.

You can then try to play with the RawUSBFeature via `spec.features`

Or the intended way is to add the command to here:
https://github.com/ap--/python-seabreeze/blob/5adee55e0fa83126e7fd264291eef7b8a9c87413/src/seabreeze/pyseabreeze/protocol.py#L40-L50

And then implement the methods here:
https://github.com/ap--/python-seabreeze/blob/5adee55e0fa83126e7fd264291eef7b8a9c87413/src/seabreeze/pyseabreeze/features/lightsource.py

The methods will then all go via `self.protocol` and its `send`, `receive` and `query` methods.

When you'll read the data sheet you'll understand what i mean.

Anyways if you need further help let me know :+1: And feel free to ask if something's unclear!

Cheers,
Andreas

moritzj29, 2020-05-07:
Hi Andreas,
thanks for your thorough explanation! This was really helpful for me! I managed to get the light source working using this protocol
`0x03: '<BH',   # set Strobe/Lamp enable Line`
and sending just 1 or 0. (Or using the equivalent raw USB commands). 

I would like to contribute this to the codebase, but I'm not sure which way to choose. As I understand, the `lightsource.py` should be the right place. But the proposed methods from the cseabreeze API do not really fit the functionality of this simple toggle switch. The methods target at more sophisticated light sources which can be controlled individually. Here, I only use the very general light on/off switch which controls the "Lamp Enable Line (J2 pin 4)". Therefore I suggest implementing this on the same level as setting the ingetration time etc. Also in the manual it is listed together with these very basic commands.
I also suggest implementing the "Query Status" method to read information from the instrument, e.g. is lamp on or not. I think this also fits into the correspondig spectrometer classes, since the returned data differs in content and encoding.

What do you think? Since I only have a "small" spectrometer, I can't compare or try the cseabreeze functionalities. So it's up to you to tell me whether this makes sense or not 😃 I see that pyseabreeze should exactly match the methods provided by cseabreeze, which makes perfect sense.

If it's too much effort for this basically 5 lines of code, that's ok with me. I got it working now, thanks to this great repository 👍

ap--, 2020-05-08:
Hi @moritzj29,

Awesome :tada: Great that you got it working!

I'd definitely say we'll integrate your changes! Best way to do this, would be for you to fork the repo, and create a new branch with your commits, and then create a pull request to master from that branch.

I have some time on the weekend, so I can have a look and move things around if needed.
Since the USB650 is not supported by cseabreeze, we don't really have to worry about compatibility for that backend. :blush: 

Cheers,
Andreas

ap--, 2020-05-22:
merged to master 8b1736492f17a378a26b7c2e0957f441799da478

ap--, 2020-05-22:
Thanks again :+1: :tada: :heart:

closed 2020-05-22 by ap--