ap--/python-seabreeze #85: usb_raw_write/read

MaxW93, 2020-05-24:
### spectrometer and system information ###

- model: NIRQUEST 512.2.5
- operating system: Windows 7 64bit
- python version: 3.7
- python-seabreeze version: current master
- installed-via: conda

### current problem ###
Hi,
I'm struggeling with the raw_usb-read and write functions. Simple functions like get_wavelength work perfectly fine. The issue has something to do how I setup my spectrometer class. I'm also not sure which endpoint to choose.
In the example below I try to query the status with 0xFE and read the output afterwards.
Documentation for the NIRQUEST spectrometer:
[https://www.photonicsolutions.co.uk/upfiles/OEM-Data-Sheet-NIRQuest.pdf]

**Code:**
`import seabreeze
seabreeze.use('cseabreeze')
from seabreeze.spectrometers import Spectrometer
spec = Spectrometer.from_first_available()
wavelengths = spec.wavelengths()
print(wavelengths)
seabreeze.cseabreeze._wrapper.SeaBreezeRawUSBBusAccessFeature.raw_usb_write('0xFE','primary_out')
reading_data=seabreeze.cseabreeze._wrapper.SeaBreezeRawUSBBusAccessFeature.raw_usb_read('primary_in')`

**Console:**
[ 878.308       884.09200799  889.89658261  895.72144588  901.56631979
........
 2249.15334375 2251.42626989 2253.65026274 2255.8250443  2257.95033656
 2260.02586153]
Traceback (most recent call last):
File "C:\Heating GUI\HighGainTest.py", line 16, in <module>
    seabreeze.cseabreeze._wrapper.SeaBreezeRawUSBBusAccessFeature.raw_usb_write('0xFE','primary_out')
TypeError: descriptor 'raw_usb_write' requires a 'seabreeze.cseabreeze._wrapper.SeaBreezeRawUSBBusAccessFeature' object but received a 'str'

ap--, 2020-05-25:
Hi @MaxW93 

what are you trying to achieve exactly?
Maybe the functionality you need might already be abstracted in another module.

Also there seems to be a few misunderstandings on how to use the feature. I'll provide an example. If you could help improving the documentation to make it more clear for others how to use it, it'd be great :+1: 

```python
import struct  # needed for packing binary data into bytestrings
from seabreeze.spectrometers import Spectrometer                                                 
                                                                           
spec = Spectrometer.from_first_available()                                                       

# features need to be accessed on the spectrometer instance via spec.f or spec.features
# the data you provide needs to packed into a bytestring via struct.pack
spec.f.raw_usb_bus_access.raw_usb_write(data=struct.pack('<B', 0xFE), endpoint='primary_out')

# when reading via the raw_usb feature you can easily block the spectrometer.
# so make sure to only read when you expect new data (after you sent a command)
# and only read as many bytes as the command returns
output = spec.f.raw_usb_bus_access.raw_usb_read(endpoint='primary_in', buffer_length=16)  

print(output)
>>> b'\x00\x08p\x17\x00\x00\x00\x00\x00\x08\x00\x00\x00\x00\x80U'
# extract by providing the data format from the documentation and `output` to struct.unpack 
```

Let me know if this makes things more clear :blush: 
Cheers,
-Andreas

MaxW93, 2020-05-25:
Thank you for your quick answer and all the work you have put into this project. 
I should have thought about translating the data to bytestring. I'm trying to activate the high gain feature. Which can be done according to the documentation with '0x0C 0x01' 
You definetly made things clear for me! I think adding this example of yours to the documentation is a great idea.

P.S. Is the set_scan_to_average available for the NIRQUEST spectrometer? You mentioned in a another issue that this feature is only available for Spark and STS spectrometers. But I can set the average in OceanView.  What is the difference betwenn these scans to average functionalities? 

Cheers Max

ap--, 2020-05-26:
Some spectrometer models use a different communication protocol that support averaging spectra on the spectrometer itself.

My guess is that for the NIRQUEST spectrometer OceanView just does spectrum averaging in post (maybe a rolling average)

commit d20c27b8 (2020-06-09): [docs] pyseabreeze: add example to raw usb access feature

Close #85.
closed 2020-06-09 by ap--